360Controller/360Controller #1043: No devices found even it is listed in USB device tree (Xbox one wireless controller)

Shawolyyx, 2020-04-05:
### Type of Controller
Xbox One

### OS Version
macOS Catalina Version 10.15.4

### Driver Version
360Controller 1.0.0-alpha.5
360Controller 1.0.0-alpha.4
360Controller 0.16.11 (360ControllerInstall_0.16.11.dmg and 360ControllerInstall_0.16.11_Notarized.dmg)

### Connection Method
Wired (micro USB)

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x02d1
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
When I connect it with my macbook pro, it vibrates. Nothing else can be observed on the controller.

### Response in Preference Pane
My controller doesn't appear in the Xbox 360 controllers preference pane. It's not detected by the driver.

### Detailed Explaination of Issue
The issue is quite simple. I tried out a few releases of the driver (successfully installed), but when I connected my controller to my macbook pro via a micro USB cable, it couldn't be detected by the driver. Nothing worked.

Shawolyyx, 2020-04-05:
My controller is listed in `Apple Menu -> About this Mac -> System Report -> Overview -> Hardware -> USB`:

Controller:

  Product ID:	0x02d1
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	2.03
  Serial Number:	7EED8B743ACF
  Speed:	Up to 12 Mb/s
  Manufacturer:	Microsoft
  Location ID:	0x14100000 / 6
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

Guatama, 2020-04-06:
Same problem :(.  Same product and vendor IDs

toastedlightly, 2020-04-07:
same issue

toastedlightly, 2020-04-07:
Driver will not enable. The driver has failed to load due to an unknown error, code: -603946986. Please report this error to the driver maintainer.

Have attempted multiple reinstalls 1.0.0-alpha.5

Guatama, 2020-04-08:
> Same problem :(. Same product and vendor IDs

The gamepad eventually worked with the latest alpha version (does not work on the release version)

bmitchinson, 2020-04-18:
Xbox One phantom white not working with alpha or latest release 🤔

EarthenGent, 2020-05-09:
Same as OP, Xbox One Elite Controller (V1), Mac Catalina 10.15.5 Beta

Vix3runt, 2020-05-15:
Same, Xbox elite controller v2, macOS 10.15.4, device can be found in the USB device tree but no response other than the vibration when plug the cable.