360Controller/360Controller #1044: USB ports stopped working for other devices, cannot completely uninstall

DmitriVincent, 2020-04-07:
Xbox One controller
iMac - Catalina 10.15.4
Driver 5
Wired connection

The controller works, but the computer no longer recognizes anything plugged into the USB ports, except for the Xbox controller. The SD reader and USB-C work fine, but the regular USBs no longer recognize anything other than the controller. They still power devices, but I cannot access anything. I reset the SMC following Apples instructions (basically power off and unplug) and they're still not working. 

I tried to uninstall the driver, but I see there are still files (360Controller.kext, Wireless360controller.kext, WirelessGamingReceiver.kext, WirelessOneController.kext) but I cannot remove them. I read on another post that theres an uninstall script but I'm not familiar with this. 

This problem started when I installed the driver so please assist. 

Thank you!

FranticRain, 2020-04-08:
If you go into the preference pane of the driver (in your System Preferences) there is an "Advanced" tab that has an uninstall button in it. Please use that. If you have already uninstalled the preference pane, please re-install the driver and then use the uninstall feature of the driver.

DmitriVincent, 2020-04-08:
Thanks for your prompt response. That's how I originally uninstalled it. I installed it and uninstalled it again using the same advance tab, but the USBs aren't working and I see there are still files when I search. The uninstall.command script is there now, but it won't allow me to run it. I did "get info" and made sure I have permission, but it still doesn't work if I double click on it. Is there another way to remove everything or do you know how to fix the USBs? My external hard drive is through the USB so I can't back anything up right now to do a reset.

FranticRain, 2020-04-08:
The driver only attaches to devices that the operating system offers it. It can't break all of your USB devices unless your macOS kernel is acting incorrectly. I cannot help you debug macOS, I don't know anything about how it works. I also can't help you debug your macOS permissions issues. There are multiple means provided to help you uninstall the driver, up to and including deleting the files themselves. Any admin on a machine should be capable of deleting these drivers. You will need to resolve your own permissions issues by yourself.

closed 2020-04-08 by FranticRain