360Controller/360Controller #1045: Are Wirless XBox 360 controller supported with 1.x?

ajmas, 2020-04-08:
### Type of Controller
Xbox 360

### OS Version
10.15.4

### Driver Version
1.0.0-alpha.5

### Connection Method
Official Microsoft wireless adapter, that came with 360 controller

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Microsoft Wireless controller
<br/>Product ID: 0x028f
<br/>Vendor ID: 0x045e

### Response of Controller
Having trouble syncing. 

### Response in Preference Pane
Preference pane shows not devices connected.

### Detailed Explanation of Issue
I am using a 16" MacBook Pro with a Anker adapter plugged in. Into that I have plugged in my Microsoft wireless adapter. I tried syncing with the steps in the following video:

https://www.youtube.com/watch?v=3o-RgAM_GUY

I just end up with a flashing green ring around the X-Box button. I have tried multiple controllers and did also look in /var/log/system.log, but see nothing of note.

Is the Wireless 360 controller meant to be supported with the driver rewrite?

FranticRain, 2020-04-08:
Nope.

closed 2020-04-08 by FranticRain

ajmas, 2020-04-08:
Is there anything on the roadmap for this and what are the challenges at this point?

FranticRain, 2020-04-08:
> Is there anything on the roadmap...

Nope. Number one priority is getting the driver ported to DriverKit, since the current framework is being deprecated.

> ...what are the challenges at this point?

Lack of interest in working on this project.