360Controller/360Controller #1048: Cannot uninstall controller

sebastianfelipe, 2020-04-11:
### Type of Controller
Xbox 360
### OS Version
Mac OSX Mojave
### Driver Version
360Controller 1.0.0-alpha.5
### Connection Method
Wireless Adapter
### Detailed Explaination of Issue
The Xbox 360 controller app still appearing in the system preferences after run the uninstall.command

FranticRain, 2020-04-11:
Have you tried using the uninstall feature in the "Advanced" section of the preference pane? And can you confirm if there are still "Xbox" related files in `/Library/Extensions`?

FranticRain, 2020-05-15:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2020-05-15 by FranticRain