360Controller/360Controller #1049: Third Party Controller not detected

saydoujah, 2020-04-11:
### Type of Controller
Xbox One - Third Party (Power A)

### OS Version
10.15.3 Catalina

### Driver Version
360Controller 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: XB1 Classic controller:
<br/>Product ID: 0x581a
<br/>Vendor ID: 0x24c6

### Response of Controller
Device is detected by the MAC but no reaction from controller. Driver app shows as 'No devices Found".

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
When I plug the wired controller the device is not detected by the driver and there is no reaction from the controller no lights, no vibration.... My Macbook detects the controller.

EarthenGent, 2020-05-09:
Same as OP, Catalina 10.15.5 Beta, Xbox One Elite Controller

FranticRain, 2020-05-15:
Added in alpha 6.

closed 2020-05-15 by FranticRain

jm2, 2020-08-13:
Why was a PowerA controller (according to the device ID) added as the Hyperkin Duke here (according to the key)?