360Controller/360Controller #1052: Xbox Elite Serie 2 Controller note detected by the driver.

Jonathanallibe, 2020-04-15:
### Type of Controller
Xbox Elite Serie 2

### OS Version
MacOS Catalina 10.15.4

### Driver Version
360Controller 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Xbox Controller Elite Serie 2
  Product ID:	0x0b00
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	4.07
  Serial Number:	3032363330303335343937303032
  Speed:	Up to 12 Mb/s
  Manufacturer:	Microsoft
  Location ID:	0x14300000 / 3
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
The controller is detected on the computer but not by the driver so no response from the controller.
The only response that I have is when I plug the controller in the controller, it vibrates and then the middle button is blinking slowly.

### Response in Preference Pane
Not Detected in the preference pane.

### Detailed Explaination of Issue
Basically, when I plug the controller it is not detected in the preference panel by the driver.
I have tried to restart my computer, uninstall and re-install the drive and swap the cable.
Thank you in advance for your help.

Jonathanallibe, 2020-04-15:
I have tried as well the build you've done in the other thread about about the elite v2.
The controller is now showing in the preference pane but no inputs are detected.

franszo, 2020-04-27:
+1 wondering if theres any progress on this one. Such a shame a pricey top tier controller has no proper support ;(

jonathan-hewo, 2020-04-27:
Hey @franszo, You need first to update the controller on a Windows to have it detected by the driver.
Now it is working well for me.

franszo, 2020-04-27:
Hey! Must be a windows PC? Mine has been ipdated on the Xbox for the latest. I’ll give it a try. Is it the latest release which working or I saw something ipdated on another thread?

Jonathanallibe, 2020-04-30:
Sorry for the late answer. I have updated mine on a windows PC. I don’t know if it changes something to update it on a Xbox... i don’t think so.
In regards to the version of the driver I have used one version found in another thread about the xbox serie 2 controller. The developer has built a specific version for it. I don’t know if the changes he has done in this specific version of the driver have been included in the latest version...

closed 2020-04-30 by Jonathanallibe