360Controller/360Controller #1053: Third party Xbox one controller not found

katbard, 2020-04-16:
### Type of Controller
Xbox One Third Party

### OS Version
Catalina 10.15.4

### Driver Version
360Controller 1.0.0-alpha.5 and alpha 4

### Connection Method
Wired

### Device Name and Info
>Device Name: XB1 Classic Controller
<br/>Product ID: 0x581b
<br/>Vendor ID: 0x24c6

### Response of Controller
Controller does not respond, it does not light up or vibrate. 360controller shows “no devices found”

### Response in Preference Pane
The controller appears in the preference pane on Mac but does not connect. It says “no devices found” all of the buttons, joysticks, and triggers do not work.

### Detailed Explaination of Issue
Third party wired Xbox one controller connected through USB, appears under “hardware” in the system report window on Mac, but does not connect to 360controller driver. Under “controller test” it is not detected nor does it vibrate or light up when plugged in. Tried unsuccessfully in both USB slots and multiple cords on both alpha 4 and 5.

katbard, 2020-04-16:
Thank you in advance, I am fairly new around here so I am not sure how to do it on my own.

jjfattz, 2020-04-27:
I have the same controller.  Would be nice to have it added.

saundemail, 2020-05-09:
I have the same controller.  Can it be added?

FranticRain, 2020-05-15:
Added in alpha 6.

closed 2020-05-15 by FranticRain

jm2, 2020-08-13:
Why was a PowerA controller (according to the device ID) added as the Hyperkin Duke here (according to the key)?