360Controller/360Controller #1054: 1.0.0-alpha.5 XBox 360 - no devices found

phnakarin, 2020-04-19:
### Type of Controller
Xbox 360

### OS Version
macOS Catalina version 10.15.4
MacBook Pro (Retina, 13-inch, Early 2015)

### Driver Version
360Controller 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
```
Vendor-Specific Device:

  Product ID:	0x028f
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	2.04
  Speed:	Up to 12 Mb/s
  Location ID:	0x14100000 / 6
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0
```

### Response of Controller
The in-the-middle Xbox button is blinking

### Response in Preference Pane
Controller test displays "No devices found".

### Detailed Explanation of Issue
I granted downloading apps from developer "Drew Mills" regarding the Readme. The installation went pretty smooth. I rebooted my Macbook and USB connected my 360 controller right after it was fully booted up. It turned out that my macOS was able to see the controller under `About this Mac -> System Report -> Overview -> Hardware -> USB -> Vendor-Specific Device` but not from `System Preferences -> Xbox 360 Controllers`.

Here is the outputs of `kextutil` command.
```
$ sudo kextutil /Library/Extensions/360Controller.kext

Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).
```
I was wondering what I did wrong. Did I miss a crucial step? This is my first time trying to set up this driver. I did try to uninstall/reinstall but no dice. Any suggestion would be very appreciated.

karlotroiano, 2020-04-19:
same issue here, follow.

germanviscuso, 2020-04-22:
Same here, using 360Controller 1.0.0-alpha.5, same output by kextutil
Macbook Pro Retina mid 2014
MacOS High Sierra 10.13.5
But in my case it's a wireless receiver:

  Product ID:	0x0291
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.07
  Speed:	Up to 12 Mb/sec
  Location ID:	0x14100000 / 9
  Current Available (mA):	500
  Extra Operating Current (mA):	0

devinrhode2, 2020-04-23:
same issue on mac os 10.15.4, same device as OP

EarthenGent, 2020-05-09:
Same issue, Xbox Elite Controller V1, Mac Catalina 10.15.5 Beta, otherwise same as OP

nh4rxthon, 2021-01-15:
Exact same issue. wired xbox 360 controller. Mac OS Mojave 10.14.4. MacBook Pro (Retina, 13-inch, Late 2013)

kenmccann, 2021-01-15:
Same issue.  Wireless Receiver (non-Microsoft).  MacOS Catalina 10.15.6.  MacBook Pro (15-inch, 2017)

panterapxviii, 2021-04-10:
someone can fox it for Mac OS Big Sur ?