360Controller/360Controller #1057: Generic S600 gamepad (A9) not recognized

inieto, 2020-04-26:
### Type of Controller
Generic S600 gamepad (A9)
Very simlilar to this one: https://www.alibaba.com/product-detail/For-T3-S3-S5-PS3-Bluetooth_62218726072.html?spm=a2700.7724857.normalList.59.4beaa8e9vVhNl6

### OS Version
Catalina 10.15.4

### Driver Version
1.0.0-alpha.5

### Connection Method
Bluetooth

### Device Name and Info
```
S600 Gamepad:
  Address:	9C-54-1C-02-08-E4
  Major Type:	Miscellaneous
  Minor Type:	Unknown
  Services:	Bluetooth Gamepad   
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Cambridge Silicon Radio (0x4, 0x1060)
  Bluetooth Core Spec:	2.1
  Firmware Version:	0x011B
  Vendor ID:	0x1949
  Product ID:	0x0402
  Class of Device:	0x00 0x00 0x0000
  AFH:	On
  AFH Map:	FFFFFFFFFFFFFFFFFF7F
  RSSI:	-38
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  Host Connectable:	No
  EDR Supported:	Yes
  eSCO Supported:	No
  SSP Supported:	Yes
````

### Response of Controller
It only pairs with bluetooth as a `S600 gamepad` (with keyboard icon)

### Response in Preference Pane
It does not recognize it. Wired also does not work (it is not even recognized in the `System Report...` menu from `About this mac` modal.

### Detailed Explaination of Issue
I've tried build from issue #1033, build "1.0.0-alpha.5", build "1.0.0-alpha.2". Also tried to every kind of combination: 
"wired" -> Led 2 ON (also charging, when 100% charged turns OFF)
"Home + X". Led 3 ON.
"Home + B". Led 3&4 ON.
"Home + Y". Led 1&3 ON.
"Home + A". Led 4 ON.

*On Home+A:*
leftBtn: a, rightBtn: d, upBtn: w, downBtn: s, selectBtn: (del), startBtn: (enter),
X: 1, Y: 4, B: 3, A: 2
left joystick (arrows), 
right joystick: left: j, up: i, right: l, down: k
LT1: 5, LT2: 7, RT1: 6, RT2: 8

*On Home+B:* (first letter is key_press, second letter is key_release)
leftBtn: aq, rightBtn: dc, upBtn: we, downBtn: xz, selectBtn: og, startBtn: lv,
X: yt, Y: uf, B: jn, A: hr
left joystick: left: aq, up: we, right: dq, down: xz  (same as arrow buttons)
right joystick: <doesn't work>
LT1: im, LT2: og, RT1: kp, RT2: lv 

*On Home+Y*: nothing works
*On Home+X*: nothing works

When Mode is set with Home+A or Home+B, the Gamepad is not recognized in the "Xbox 360 controllers" menu on System preferences

inieto, 2020-04-26:
```
$ sudo kextutil /Library/Extensions/360Controller.kext
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).
```

FranticRain, 2020-05-15:
This controller is not supported by official Xbox hardware. As such, its support is outside the scope of this driver. We cannot and will not support this controller.

closed 2020-05-15 by FranticRain