360Controller/360Controller #1060: Driver does not recognize the controller

Ungamerahi, 2020-05-03:
### Type of Controller
Xbox One

### OS Version
macOS Mojave Version 10.14.6

### Driver Version
360Controller 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: (does not appear in the report, should it be Xbox One controller?)
<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
The controller lights up and even vibrates a little bit when connected.

### Response in Preference Pane
The driver does not recognize the controller. It still says "No devices found" Nothing happens at all. 

### Detailed Explaination of Issue
I was playing Fortnite but when I connected the controller, everything went nuts. The buttons are wrongly mapped. The "Y" works as "RB", the "A" button works as the "B", etc. 
I googled how to fix it and discovered this amazing driver you created, but following every step, I wasn't able to fix my problem because the driver does not recognize the controller. The laptop does recognizes it because it appears in the System Report you ask us to look for. 
I hope this can be solved, I would be so happy. Thanks in advance, mates. 

[image]

jbud, 2020-05-05:
same here with wireless controller, same OS same version.

EarthenGent, 2020-05-09:
Same here with Catalina 10.15.5 Beta, Xbox One Elite Controller

colmlg, 2020-05-09:
Same here, mac os 10.14.6 Xbox 360 Controller Driver Version 1.0.0-alpha.5

colmlg, 2020-05-09:
Just went to the `Advanced` tab in the 360 driver pref pane and `Enable Driver` was unticked somehow. I definitely did not do that as I did not know the pref pane existed until now. Ticked it again and the controller is picked up again, weird.