360Controller/360Controller #1063: I get the error installation failed

ninja8170, 2020-05-11:
Hello trying to install on catalina but any version i have tried i get instillation failed , i have tried disabling gatekeeper but still same error 
### Type of Controller
Xbox One

### OS Version
10.15.4 
### Driver Version
Alpha 5
### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

ninja8170, 2020-05-11:
This is the log 

May 11 16:23:28 Dwaynes-MBP Installer[96458]: @(#)PROGRAM:Install  PROJECT:Install-1000
May 11 16:23:28 Dwaynes-MBP Installer[96458]: @(#)PROGRAM:Installer  PROJECT:Installer-965.1
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Hardware: MacBookPro15,3 @ 2.90 GHz (x 12), 32768 MB RAM
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Running OS Build: Mac OS X 10.15.4 (19E287)
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: USER=admin
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: COMMAND_MODE=unix2003
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: DISPLAY=/private/tmp/com.apple.launchd.ge9ezeN6hg/org.macosforge.xquartz:0
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: PATH=/usr/bin:/bin:/usr/sbin:/sbin
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: LOGNAME=admin
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.9m6MbqUEDz/Listeners
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: HOME=/Users/admin
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: SHELL=/bin/zsh
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: TMPDIR=/var/folders/mk/49zykld57xs70hyp9lm_vd140000gq/T/
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: __CF_USER_TEXT_ENCODING=0x1F7:0x0:0x2
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: XPC_SERVICE_NAME=com.apple.xpc.launchd.oneshot.0x100000c5.Installer
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Env: XPC_FLAGS=0x0
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Install360Controller  Installation Log
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Opened from: /Volumes/360ControllerInstall/Install360Controller.pkg
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Package Authoring Error: <background_scaling> has an unsupported MIME type: X-NSObject/NSNumber
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Package Authoring Error: <background_alignment> has an unsupported MIME type: X-NSObject/NSNumber
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Package Authoring Error: <layout-direction> has an unsupported MIME type: X-NSObject/NSNumber
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Failed to load specified background image
May 11 16:23:28 Dwaynes-MBP Installer[96458]: Product archive /Volumes/360ControllerInstall/Install360Controller.pkg trustLevel=350
May 11 16:23:28 Dwaynes-MBP Installer[96458]: External component packages (1) trustLevel=350
May 11 16:23:43 Dwaynes-MBP Installer[96458]: Administrator authorization granted.
May 11 16:23:45 Dwaynes-MBP Installer[96458]: ================================================================================
May 11 16:23:45 Dwaynes-MBP Installer[96458]: User picked Standard Install
May 11 16:23:45 Dwaynes-MBP Installer[96458]: Choices selected for installation:
May 11 16:23:45 Dwaynes-MBP Installer[96458]: 	Install: "Install360Controller"
May 11 16:23:45 Dwaynes-MBP Installer[96458]: 	Install: "Install360Controller"
May 11 16:23:45 Dwaynes-MBP Installer[96458]: 		Install360Controller.pkg#Install360Controller.pkg : com.mice.pkg.Xbox360Controller : 1.0.0-alpha.5
May 11 16:23:45 Dwaynes-MBP Installer[96458]: ================================================================================
May 11 16:23:45 Dwaynes-MBP Installer[96458]: It took 0.00 seconds to summarize the package selections.
May 11 16:23:45 Dwaynes-MBP Installer[96458]: -[IFDInstallController(Private) _buildInstallPlanReturningError:]: location = file://localhost
May 11 16:23:45 Dwaynes-MBP Installer[96458]: -[IFDInstallController(Private) _buildInstallPlanReturningError:]: file://localhost/Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg
May 11 16:23:45 Dwaynes-MBP Installer[96458]: Set authorization level to root for session
May 11 16:23:45 Dwaynes-MBP Installer[96458]: Will use PK session
May 11 16:23:45 Dwaynes-MBP Installer[96458]: Using authorization level of root for IFPKInstallElement
May 11 16:23:45 Dwaynes-MBP Installer[96458]: Starting installation:
May 11 16:23:45 Dwaynes-MBP Installer[96458]: Configuring volume "Macintosh HD"
May 11 16:23:45 Dwaynes-MBP Installer[96458]: Preparing disk for local booted install.
May 11 16:23:45 Dwaynes-MBP Installer[96458]: Free space on "Macintosh HD": 121.8 GB (121799065600 bytes).
May 11 16:23:45 Dwaynes-MBP Installer[96458]: Create temporary directory "/var/folders/mk/49zykld57xs70hyp9lm_vd140000gq/T//Install.96458rHbJHC"
May 11 16:23:45 Dwaynes-MBP Installer[96458]: IFPKInstallElement (1 packages)
May 11 16:23:45 Dwaynes-MBP installd[1289]: PackageKit: Adding client PKInstallDaemonClient pid=96458, uid=503 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)
May 11 16:23:45 Dwaynes-MBP Installer[96458]: PackageKit: Enqueuing install with framework-specified quality of service (utility)
May 11 16:23:45 Dwaynes-MBP installd[1289]: PackageKit: ----- Begin install -----
May 11 16:23:45 Dwaynes-MBP installd[1289]: PackageKit: request=PKInstallRequest <1 packages, destination=/>
May 11 16:23:45 Dwaynes-MBP installd[1289]: PackageKit: packages=(
	    "PKLeopardPackage <id=com.mice.pkg.Xbox360Controller, version=1.0.0-alpha.5, url=file:///Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg>"
	)
May 11 16:23:45 Dwaynes-MBP installd[1289]: PackageKit: Set reponsibility for install to 96458
May 11 16:23:45 Dwaynes-MBP installd[1289]: PackageKit: Extracting file:///Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg (destination=/Library/InstallerSandboxes/.PKInstallS [... cut ...]
May 11 16:23:47 Dwaynes-MBP Installer[96458]: Install failed: The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.
May 11 16:23:47 Dwaynes-MBP Installer[96458]: IFDInstallController 1BBF480 state = 8
May 11 16:23:47 Dwaynes-MBP Installer[96458]: Displaying 'Install Failed' UI.
May 11 16:23:47 Dwaynes-MBP Installer[96458]: 'Install Failed' UI displayed message:'The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.'.

FranticRain, 2020-05-15:
Looks like you aren't an admin on your machine. This is not a bug with the driver. You need to be a machine admin to install the driver. There's nothing we can do about that.

closed 2020-05-15 by FranticRain