360Controller/360Controller #1068: Error: code: -603946986

robhoward83, 2020-05-15:
Xbox 360 

Os 10.11

Version 6

Wireless 

Gives an error code when activating the driver

I have the latest driver - version 6, which this issue was supposed to fix (according to other tickets)

However, I'm still getting the error: code: -603946986 when I click to activate the driver.

FranticRain, 2020-05-15:
Can you confirm if this build resolves the error code? The fix must have gotten lost in the hiatus. 
[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/4637358/360ControllerInstall.dmg.zip)

robhoward83, 2020-05-16:
Nope, I just reinstalled. Still giving me the same error code.

FranticRain, 2020-05-17:
Of course it's not the easy. How about this one?

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/4639477/360ControllerInstall.dmg.zip)