360Controller/360Controller #1070: 360Controller 1.0.0-alpha.6 is not notarized and contains no changes

silentj, 2020-05-17:
### Type of Controller
Xbox 360

### OS Version
10.15.4

### Driver Version
1.0.0-alpha.6 (reverted to 1.0.0-alpha.5 and the problem disappeared)

### Connection Method
Wired

### Device Name and Info
N/A

### Response of Controller
Controller not found

### Response in Preference Pane
Unable to enable the driver

### Detailed Explaination of Issue
I tried to upgrade from 1.0.0-alpha.5 to 1.0.0-alpha.6 and was unable to enable the new driver or get any response from the controller. When I clicked "Enable Driver", it took me to the Security preferences but there was no prompt to allow the extension. 

Whilst looking for answers, I first noticed that alpha.6 says it's built from the same commit as alpha.5 (`a6bc94e` - currently the latest commit on `master` in the repo).

I also tried the command mentioned in [this comment](https://github.com/360Controller/360Controller/issues/989#issuecomment-565855598), which told me the driver wasn't notarized.

Uninstalling alpha.6 and reinstalling alpha.5 worked fine for me (alpha.5 reports it _is_ notarized).

FranticRain, 2020-05-17:
Alpha 6 is based off of the `modern-rewrite` branch, which does contain changes. I messed up and started tracking master with alpha 5, and there’s no way to edit a release to change the branch it is tied to, so that’s how it is. I’ll look into the notarization and get back to you.

FranticRain, 2020-05-17:
Release has been updated with a binary that is signed/notarized. Tags have been updated to track the right commits. Unfortunately, it looks like the releases are still comparing to `master`. Not sure there is anything I can do about that.

closed 2020-05-17 by FranticRain