360Controller/360Controller #1071: Deleting driver completely to simply use standards driver in MacOS 10.15

ozjsean, 2020-05-17:
### Type of Controller
Xbox One

### OS Version
10.15.4

### Driver Version

### Connection Method
Wireless (Bluetooth)

### Device Name and Info
N/A

### Response of Controller
Recognize controller on Mac, key mapping faulty

### Response in Preference Pane

### Detailed Explanation of Issue
Simply want to use the standard Xbox Driver in MacOS 10.15 to use the controller on Bluetooth but your driver which I previously downloaded may be interfering this. I have uninstalled your driver but some of the kext are still there, I'm assuming that this is the core of the problem.
1. Is this assumption correct?
2. If so, which kext do I uninstall?
3. How do I do this? Neither 'sudo kextunload' nor 'sudo rm -rf' worked.

Under
> /Library/StagedExtensions/Library/Extensions
there are files like
>360Controller.kext, Wireless360Controller.kext, WirelessGamingReceiver.kext, WirelessOneController.kext
which seem suspicous.

Thank you in advance!

ozjsean, 2020-05-17:
I've managed to deleted the kernel files using this method
https://nektony.com/how-to/remove-kext-on-mac

The deleted kernel files are
'360Controller.kext', 'Wireless360Controller.kext', 'WirelessGamingReceiver.kext', 'WirelessOneController.kext'

The issue still continues. Do you know how to fix this?

FranticRain, 2020-05-17:
The uninstaller in the app and the uninstaller included in the latest release assets are the approved ways to delete the driver. If you’ve removed those kexts from your machine and rebooted, then the driver is in no way responsible for the issues you are having.

Not to mention the fact that the driver doesn’t interfere with Bluetooth connections at all, and cannot and will not cause a problem for Bluetooth connections to macOS. 

Furthermore, issues with incorrect mapping of buttons would be a problem either with the game you are using, or with macOS. There’s no standard button mapping in macOS, so it’s just using the buttons assigned by the Bluetooth HID report descriptor. It is up to individual games to support that mapping.

closed 2020-05-17 by FranticRain

ozjsean, 2020-05-17:
Thanks FranticRain
I'm trying to play Fortnite via EpicGames Launcher (ver 10.16)
If anyone else has a issue, would be helpful to share, thank you

FranticRain, 2020-05-17:
There are many people who have issues with Fortnite. It simply doesn't support controllers very well on macOS.

ozjsean, 2020-05-17:
Yes, see many other similar issues raised although none of them seem to have good answers. Weird how it’s working fine on some and not on others. Guess I just have to wait for Epicgames to update or MacOS 16 some how solving the issue...?!

k-funk, 2020-10-14:
@ozjsean were you able to resolve your issue? If not, does your controller work correctly in this site? I'm having the same issue. Unsure if it's related to this package or not, but I have tried my controller on 4 devices:
* xbox one
* 2015 macbook air
* 2019 macbook pro
* 2017 imac (this is the only one that has problems, and this is the only one that had 360Controller on it)

I also have a visual issue that the icon shows a bluetooth icon instead of a gamerpad when looking in bluetooth preferences. Whenever that happens, I know it's not going to work at all. If the gamerpad icon shows up, it will partially work for me.