360Controller/360Controller #1075: Wrong version 5 and 6

mac4kent, 2020-05-24:
### Type of Controller
360 Controller wired

### OS Version
MacOS Catalina 10.15.5 beta

### Driver Version
N/A

### Connection Method
Wired USB

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
N/A

### Detailed Explaination of Issue
N/A

I found https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.6 but wrong version to be alpha 5 on my system preferences. Need to fix match version see picture
[image]

bgruszka, 2020-05-24:
Agree, I have the same

mac4kent, 2020-05-24:
True

FranticRain, 2020-05-24:
Updated release to version that fixes this. Re-download the release to resolve the issue.

closed 2020-05-24 by FranticRain

mac4kent, 2020-05-24:
I check still 5 not 6. please send me link make sure match 6 not 5 on System Preferences @FranticRain

FranticRain, 2020-05-24:
You're right. Xcode was being a jerk and didn't save my changes to the `.xib`. I've fixed it and re-uploaded it to the release page. Give it another shot.

mac4kent, 2020-05-24:
@FranticRain  hehe ok. When you are finish everything update let me know.

FranticRain, 2020-05-24:
It should be uploaded correctly now. I've tested it to confirm.

mac4kent, 2020-05-24:
Good let me inspect.

mac4kent, 2020-05-24:
it fixed congratulation good job.
Now I can see 6 there correct ya  ayayayayyyyy 

Now are you still work on left vibrate and right vibrate? @FranticRain

FranticRain, 2020-05-24:
No. Apple broke the way we’re implementing force feedback. It will not work anymore with this driver.

mac4kent, 2020-05-24:
Oh ok I got it understand. I can do force to Apple developers about xbox 360 controller USB connect wire.