360Controller/360Controller #1077: Future of this repo

garrrettt, 2020-05-28:
In the latest release, you (@FranticRain ) said that it would be your last time to touch the code. Does this mean that you are leaving it up to the community to maintain this repo in the future?

You also mentioned that this repo will need to be fundamentally rewritten using the new DriverKit (since legacy kernel extensions will no longer be supported after MacOS Catalina). Are you planning to work on this re-write? If not, is there anything we (the community) need to know in order to successfully do the re-write ourselves if we want?

FranticRain, 2020-05-28:
Kexts will be deprecated in future versions of macOS. DriverKit is the new standard. I’m working on DriverKit right now, slowly but surely. I’ve been in an out of the hospital lately due to some health stuff, but I’m working on the replacement for this repo. This repo will continue to exist for people who aren’t using versions of macOS that support DriverKit.

garrrettt, 2020-05-28:
Okay, hope your health improves. I (along with many others) definitely appreciate the work you've done on this repo. I might be interested in helping you with the DriverKit project since it's something that's useful for me and because I'm a student so I need to show some experience on my resume. Problem is, I've only used high-level languages before like Java, Dart, and JavaScript, so I've never dealt with manual memory management and such. Would you be interested in help?

FranticRain, 2020-05-28:
The USBDriverKit documentation is poor and relies on an already existing knowledge of kexts. I’ll likely be the only one with the knowledge to get that off the ground. I’ll let you know when opportunities become available. But with the way the DriverKit provisioning profiles work, you probably wouldn’t be able to build the driver, even if you had the source.

garrrettt, 2020-05-28:
Ah okay, understood. Well best of luck then!

closed 2020-05-28 by garrrettt