360Controller/360Controller #1080: Not detecting XB1 Elite Series 2

mrlukey676, 2020-05-31:
### Type of Controller
Xbox One Elite Series 2

### OS Version
10.15.3

### Driver Version
Version 1.0.0-alpha.6

### Connection Method
Wired

### Device Name and Info
N/A

### Response of Controller
The controller is detecting power as the charge button lights up.

### Response in Preference Pane
Not being detected by the software.

### Detailed Explaination of Issue
Not being detected by the software.

eksuri, 2020-06-03:
Same situation, I am able to detect an "Xbox One Wired Controller" (Xbox One S controller) wired but not Elite Wireless Controller Series 2. Firmware 4.8.1908.0.

ajkaill, 2020-07-09:
Any word on detecting Elite series 2?