360Controller/360Controller #1081: 360 wireless + receiver, several 0.16 releases, and it still won't sync

tturan1907, 2020-06-03:
First of all, thanks for all the work here. Couldn't believe when I saw the author is still responding to recent issues. Great thanks for that. Yet my Xbox360 Wireless controller doesn't connect whatever I do. 

### Controller:
Xbox360 Wireless Controller

### Mac:
OS 10.12.6

### Drivers:
0.16.5
0.16.4
0.16.2
0.16
(also tried several 1.0 versions just for the chance)
also tried with "Enable driver" on/off

### Connection Method:
Wireless Adapter 
Product ID: 0x02a9
Vendor ID: 0x045e (Microsoft Corporation)

### Response of Controller:
*Controller does not connect*
("detailed" column below for details)

### Response in Preference Pane:
Doesn't appear on the GUI at all 
("No devices found")

### Detailed Explanation of Issue:
- I plug in the Wireless Receiver/Adapter, and turn on the 360 Controller. 
- I press the connection button on the Receiver, and hold down the connection button on the Controller. 
- The light on the controller spins once (or twice or a few times sometimes). 
- Then all four lights around the circle flash on and off **quickly** for 4 times. (this is still the same even if I don't press the button on the Receiver.)
- Then the lights keep flashing forever, but now slowly after the 4 quick flashes. (the receiver light becomes solid green).

solomani, 2020-06-26:
Same, tagging this thread to monitor.

tturan1907, 2020-07-04:
@solomani check this out if you haven't already: https://github.com/360Controller/360Controller/issues/359