360Controller/360Controller #1084: macOS 11 Big Sur beta - Controller not detected

NotTheVacuum, 2020-06-23:
This is just a warning to others; it should be expected at this stage that this would be the case. I am not surprised, nor expecting any kind of update/resolution.

### Type of Controller
Confirmed with Elite Series 1

### OS Version
macOS 11 Big Sur

### Driver Version
alpha 6

### Connection Method
Wired

### Device Name and Info
0x02e3
0x045e 

### Response of Controller
Vibrates, no light

### Response in Preference Pane
Controller not recognized in pref pane

### Detailed Explaination of Issue
Controller vibrates when connected, but doesn't show up in the pref pane.

NotTheVacuum, 2020-06-23:
Just for a little more context on this (again, no expectations) - I have removed and re-installed, and checked for "Enable Driver" in the pref pane. It is unchecked, and will not stay checked. Inconsistently, you can seemingly trigger the "Blocked Extensions" dialog by toggling this, and get the option to enable several extensions (forcing a reboot) but it does not have any effect.

NotTheVacuum, 2020-06-23:
One last note for now: it's entirely likely a future beta will help. You can check the box to Enable Driver and wait a minute, and you'll get a prompt to enable extensions. These are shown - I think several are related to 360Controler
[image]

FranticRain, 2020-06-23:
USB kexts aren't supported in Big Sur. 360Controller is macOS 10 compatible only. None of those are related to the driver, and appear to be the kexts built in to macOS.

FranticRain, 2020-06-23:
Correction: Kexts need to be re-compiled at least to work on Big Sur. Since we haven't been able to compile on the latest version of Xcode for years, it's not viable to upgrade the kext to Big Sur. Instead, I'll be focusing on DriverKit support. That work is in-progress.

NotTheVacuum, 2020-06-23:
Very cool, wasn't sure what the plans were, but Driver Kit is for sure the future. Glad to hear that!

osum4est, 2020-08-07:
Interestingly enough this seems to be working fine for me on Big Sur dev beta 4 with an Xbox Elite Series 1. I'm using alpha 6 of the driver. The install says it failed but it seems to work regardless after clicking allow in Security & Privacy. 
[image]

NotTheVacuum, 2020-08-07:
Interesting indeed, I was expecting this to stay broken based on changes to USB kexts for Big Sur. I would be concerned that they may break again by (or in) GM, until the DriverKit work is done.

fdolbec123, 2020-08-07:
I tried with my Xbox 360 controller (a guitar hero controller) is not working tho. It was working in Catalina.

jakeb5271, 2020-08-23:
xbox one s controller not working in Big Sur. Any fix?

henrycc265, 2020-09-17:
> Correction: Kexts need to be re-compiled at least to work on Big Sur. Since we haven't been able to compile on the latest version of Xcode for years, it's not viable to upgrade the kext to Big Sur. Instead, I'll be focusing on DriverKit support. That work is in-progress.

I cant wait for your work to complete 
agree, as the developer from apple replied , they also suggest on working on  Apple-provided Game Controller framework,

Thanks for your feedback. Some of these Apple Arcade games are built using game engines that access game controllers in a different way than through the Apple-provided Game Controller framework, which knows how to work with the Elite V2 controller. Over time as games and game engines update we expect they’ll begin supporting more controllers. We’re regret the inconsistency of this when playing Arcade titles on macOS.

eliezedeck, 2020-10-16:
Confirmed working on Beta 10. This has also been the case above.

You only need to accept the System Extension from the **System Preferences** > **Security & Privacy** page, then reboot when asked.

Sn00perton, 2020-11-13:
> You only need to accept the System Extension from the **System Preferences** > **Security & Privacy** page, then reboot when asked.

This option does not appear for me in Big Sur public release. Has anyone got their Xbox One controller to work properly with Big Sur Released 12Nov20? I can pair by bluetooth but the keybindings are incorrect

maxpou, 2020-11-13:
I've nothing on `System Preferences > Security & Privacy`.

I upgraded my Mac to Big Sur and my Xbox Elite Controller 2 is still not fully recognized: everything works except the triggers LT/RT.

Genos72, 2020-11-21:
Same problem, Xbox 360 controller with 1.0.0-alpha.6 driver not working on Big Sur but interestingly I found the following website and it could detect my controller without loading any driver. I just load up the page using Chrome, plug-in my controller to the USB port and my controller shows up and able to detect all the button pressed and joystick. Not sure if the information will help in the development of a new driver.

https://gamepad-tester.com
https://gamepad-tester.com/for-developers

daiken002, 2020-11-28:
Interestingly whilst I have not been able to get 360Controllers to detect a device on Big Sur, I am able to use it.

I paired my Xbox Series X controller to my Mac via Bluetooth and when trying to configure it in OpenEmu it listed the controller as "unknown". I then installed 360Controller Version 0.16.11 and rebooted.

Now OpenEmu correct lists the Xbox Controller and I am able to configure the buttons (and triggers) correctly

I have just tested in a game and it works perfectly.

Not sure about any of the what and why, but I am happy that my use case is working.

[image]

rotgot32, 2021-01-02:
Still no fix yet but I found a solution to enable the kext.

First download the tool Kext-Droplet :

https://github.com/chris1111/Kext-Droplet-Big-Sur 

Then install it and reboot in recovery mode by pressing Command + R, go to terminal and deactivate the SIP with the command : csrutil disable
[image]

Reboot and load Kext Droplet, go to /Library/Extensions and drop all the "Controller" files.
[image]

You'll be able to enable the plugin after that. 
[image]

jacobmilli10, 2021-01-09:
> Still no fix yet but I found a solution to enable the kext.
>
> First download the tool Kext-Droplet :
>
> https://github.com/chris1111/Kext-Droplet-Big-Sur
>
> Then install it and reboot in recovery mode by [...]

This worked for me! Thank you

mscammell, 2021-01-18:
> This worked for me! Thank you

testerfruit, 2021-01-18:
> Still no fix yet but I found a solution to enable the kext.
>
> First download the tool Kext-Droplet :
>
> https://github.com/chris1111/Kext-Droplet-Big-Sur
>
> Then install it and reboot in recovery mode by [...]

Does this work for the macOS Big Sur with elite 2 controller with steam??!!

jacobmilli10, 2021-01-18:
> Does this work for the macOS Big Sur with elite 2 controller with steam??!!

not sure, I use a guitar hero wired controller. but shouldn't the elite 2 work natively with Big Sur?

testerfruit, 2021-01-18:
Yes the macOS natively support it and it works inside steam Big picture mode but not in the games

ja0b, 2021-01-30:
Hi there,

I'm trying to use @rotgot32 workaround but for some reason the files under /Library/Extensions are not there, any change any of you guys can zip them and shared them with me ?

Thanks!

rotgot32, 2021-01-30:
Did you install the last alpha version of Controller360 before checking the files ?
If you haven’t, install it and ignore the install failed message. Then go check on your files under /Library/Extensions again.

https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.6

ja0b, 2021-01-30:
@rotgot32 yup I had to install it a few times and restart to get these files there, I followed your instructions but this is still not working for me, I can check the "Enable Driver" checkbox, but this change is not saved so if I click another tab and then click back the option is unchecked. I'm using Big Sur 11.1 (20C69).

I'll give it another try from scratch, in the meantime any other suggestions ?
Thanks!

rotgot32, 2021-01-30:
This is weird, I am using Big Sur 11.0.1 and it works for me with the tool kext droplet as I explained on the other comment. Maybe kext droplet is not working with your version of mac os

ja0b, 2021-01-30:
@rotgot32 This one worked for me https://github.com/360Controller/360Controller/issues/1139 I'll add a comment on the thread so google searches can find it better. Thanks for your help.

testerfruit, 2021-04-23:
> Correction: Kexts need to be re-compiled at least to work on Big Sur. Since we haven't been able to compile on the latest version of Xcode for years, it's not viable to upgrade the kext to Big Sur. Instead, I'll be focusing on DriverKit support. That work is in-progress.

Hello it’s been almost a year 
How’s the driver kit support going ? 

That could solve all the problem for Xbox controller on all steam games 

it’s crazy that this thread is the only legit place about macOS controller on the internet