360Controller/360Controller #1089: Driver installs but has no effect in Catalina 10.15.5

herojig, 2020-07-20:
### Type of Controller
Original Xbox, Xbox 360, Xbox One

### OS Version
10.15.5

### Driver Version
All from v2 onward

### Connection Method
 Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Inno GamePad 
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Nothing, not recognized

### Response in Preference Pane
no device found

### Detailed Explaination of Issue
Does not act as a proper driver in this version of MacOS. Worked great in prior versions on the same machine / controller.

AndrewReitz, 2020-07-22:
Seeing the same issue on 10.15.6 w/ Wired Microsoft 360 Controller. The LED ring on the controller blinks after installing the driver but is not recognized in the preferences.

OhNotAnotherUserName, 2020-08-01:
Exactly same issue here as OP. Hoping for a solution...!

GutierrezDev, 2020-08-01:
Same issue here. Works some times with lower versions but keep stopping disconnecting.

GutierrezDev, 2020-08-01:
Same problem here.

md-siam, 2020-08-10:
Replace that with alpha_6 release.
https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.6

herojig, 2020-08-10:
@md-siam that rocks MD! I live in Nepal, have no chance of getting new hardware, so NOW I can play games with the old. Lifesaver. Thx again for doing this, I was so damned bored...

md-siam, 2020-08-11:
@herojig I am using macOS 10.15.6 & alpha_6 release works smoothly.

GutierrezDev, 2020-08-11:
im using https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.6 but the controller works and then stop working. I use with Dolphin Emulator and while playing, stops. Needs to dettach and attach the usb.