360Controller/360Controller #1099: Xbox 360 Catalina

akv12, 2020-08-21:
My Xbox controller with the wireless USB adapter. Isn't working. Dan you please help?

### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

avinoamsn, 2020-12-15:
You'd have to provide more information if you want anyone to look in to your issue. Use the issue template in your post above as a guide.

ElDuderino777, 2020-12-22:
Hi,

The reason I can’t is stated below.
There is no product or vendor info because my Mac does not recognize the
controller is plugged in. Therefore, I can’t give any more info from the
usb tree section. I need to know how to bypass the Mac not recognizing that
the controller is plugged in.

avinoamsn, 2020-12-24:
@ElDuderino777 it looks like your issue is https://github.com/360Controller/360Controller/issues/1129. I'm not sure if these issues are related, but nonetheless your issue report looks good to me–hopefully one of the maintainers is able to help you.