360Controller/360Controller #1103: 3rd Party Xbox 360 Controller Not Showing in Preference Pane

cabbagelover25, 2020-09-05:
### Type of Controller
Xbox 360

### OS Version
Cataliona
### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: VOYEE Wired Controller
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e

### Response of Controller
Green light flashing

### Response in Preference Pane
Does not show up in preference pane

### Detailed Explaination of Issue
Just does not show up when the controller is connected in the Systems Preference Pane

https://www.amazon.com/VOYEE-Controller-Compatible-Microsoft-Windows/dp/B07WD4TSNQ

jslee7897, 2020-09-05:
That Product ID and vendor ID works for me; I am using 1.0.0-alpha.6

lifechain2, 2020-09-05:
That Product ID and vendor ID works for me; I am using 1.0.0-alpha.6

wuwb, 2020-10-08:
i got No devices found.
when i checked Enable Driver in Advanced panel, an alert dialog shows,
The driver has failed to unload due to an unknown error, code: -603979775. 
Please report this error to the driver maintainer.