360Controller/360Controller #1109: Xbox One controller not connecting to Mac Big Sur

Archit1134, 2020-09-25:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

Archit1134, 2020-09-25:
I game quite a lot and I recently got a new Mac and moved from Catalina to Big Sur and I would be great if I can connect my xbox one controller to Mac Big Sur. I installed the latest driver and it worked 100% on Catalina but it isn't working out on Big Sur. 

Let me know what can be done.
Thanks!

githerder, 2020-09-27:
Any game with inherent controller support is picked up by macOS anyways, making a driver sorta not necessary. Otherwise, I'd tell ya to take a look at DriverKit documentation. This driver doesn't work on Big Sur, probably because Apple moved away from letting drivers have kernel access, instead bumping them to DriverKit. I'll link my github repo here once I've sorted out all my issues. Since 10.11 macOS has picked up controllers for games with controllers inherently supported. Your Xbox One controller should connect by bluetooth and be picked up by the system as a bluetooth device regardless of having this driver or not. (My Xbox One controller works fine in games with inherent controller support (BL2, Counterstrike)). Why one would expect a driver from a drastically different on the inside release would work on a beta that changes the very essence of the OS kinda bamboozles me. Xbox wireless controllers have been supported natively since macOS10.11 as they are treated as Bluetooth devices, not USB devices. Due to Xbox wireless controllers being supported natively and the Dev here saying they're dropping support for this driver, I doubt very much that this will end up resolved. 

tl;dr Get a wireless controller, use it in games with inherent controller support, otherwise, read about DriverKit and write a simple driver.

githerder, 2020-10-03:
Actually, any game with full controller support will support a bluetooth connected controller. Any Apple Arcade game too... The kexts exist in Big Sur. The wired devices need separate drivers, drivers that do not natively exist in Apple code. Bluetooth devices are treated differently, macOS recognizes bluetooth xbox and playstation controllers natively without any sort of driver. I know wired controllers are cheaper, but given that this driver is dead, and I share the Dev's opinion on working for free, I'd really recommend you spend the $60 on a wireless xbox one controller (or playstation wireless controller). I hacked around a bit in Driverkit, but got bored and didn't feel like crosstesting it for release. 

Apple pretty clearly states that with GameController Xbox Wireless and Playstation are supported, inputs accessible through GCController’s GCExtendedGamepad

marnovo, 2020-10-12:
Looks like a duplicate of #1084.

damian0815, 2020-11-28:
i've just got an M1 mac and i'm motivated to get wired support working. @githerder are you willing to share your code?

mwrynn, 2021-11-16:
> i've just got an M1 mac which comes with Big Sur preinstalled and i'm motivated to get wired support working. @githerder are you willing to share your code?

Hi @damian0815 - I am currently about to buy myself some Bluetooth controllers to work around this problem, but I saw your comment from about a year ago, that you might try to solve this problem yourself. Is there any chance you had any success? Thanks!

damian0815, 2021-11-16:
> Hi @damian0815 - I am currently about to buy myself some Bluetooth controllers to work around this problem, but I saw your comment from about a year ago, that you might try to solve this problem [...]

nope, no luck, sorry. turns out PS4 controllers work out of the box, so that was my "solution" haha

mwrynn, 2021-11-16:
> nope, no luck, sorry. turns out PS4 controllers work out of the box, so that was my "solution" haha

Haha, ok, thanks anyway!

letalvoj, 2021-12-04:
Solution which worked for me: https://github.com/360Controller/360Controller/issues/1180#issuecomment-986002973

First install `0.16.11`, make sure you get to the point where the settings appear in the system, yet it says no device. Then install `1.0.0-alpha6`.