360Controller/360Controller #1112: After installing, my mac wont even boot!

YourBoiNeedsHelp, 2020-10-13:
### Type of Controller
N/A

### OS Version
macOS Big Sur Beta

### Driver Version
1.0.0 Alpha 6

### Connection Method
N/A

### Device Name and Info
MacBook Air 11 inch 2014 w/ base specs

### Response of Controller
N/A

### Response in Preference Pane
I can’t access the preference pane.

### Detailed Explaination of Issue
Ever since i've installed the driver, my mac won’t boot up. When I do boot it up, it flashes the apple logo for 1-2 seconds and then restarts, and that happens over and over again. I put it in verbose boot mode and the last paragraph says:

** In Memory Panic Stackshot Succeeded ** Bytes Traced 11183 (Uncompressed 32288) **
Attempting system restart...
IOPlatformPanicAction -> AppleAHCIDiskDriver
IOPlatformPanicAction -> AppleSMC
IOPlatformPanicAction -> AppleAHCIDiskDriver
IOPlatformPanicAction -> AppleSMC

YourBoiNeedsHelp, 2020-10-15:
I tried removing all the driver files, but that didn’t work. So now I reinstalled macOS from scratch which is really annoying because I have to lose all my data, but now it works. I still won’t close this issue as macs crashing is still an ongoing problem.

AWare, 2020-10-16:
Also getting panics off latest stable on Mojave on multiple Macs, able to eventually resolve by resetting pram and SMC and uninstalling.

arnaudmm, 2020-10-20:
Damn I got kind of same issue, I am able to enter my password but then It can not open the session, the loading bar is not ending :(

Any idea how to fix this?