360Controller/360Controller #1117: Xbox One Elite works on my iMac but not my Macbook Pro

R69NiX, 2020-11-11:
### Type of Controller
Xbox One Elite

### OS Version
10.15.7 latest catalina

### Driver Version
Version 1.0.0-alpha 6

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:
I have a Macbook Pro with only  USB-C so I'm using an adapter, I have two of these all in one type adapters and it doesn't work on both. Nothing shows up in the USB tree thing.

### Response of Controller
It vibrates when first plugged in, then when I press the home button it flashes like its looking for a connection, it doesnt stay on.
### Response in Preference Pane
No controller detected.

### Detailed Explaination of Issue
Like I said this same controller and wire and driver version work on my iMac, but with my Macbook it just doesn't detect anything. I'm thinking it must be due to my adapters somehow, but I'm sure there are many people who have a new macbook with an adapter, is it known that controllers don't work with these adapters or something? I guess my problem might be hard to figure out due to the adapter and how many different kinds there are. But all other USB devices work like normal on them. They're the sort of hub type adapters that you attach to both USB-C ports on the macbook and it has a card reader, 2xUSBs and 2xUSB-C and a HDMI port.

Like I said due to the adapter and how many different types there are I appreciate that you guys might not be able to help me due to not knowing about every single adapter out there, so any help would be much appreciated 👍 

Cheers, Liam

hrdlickamartin, 2020-11-11:
I have the same problem with the same configuration

matthax, 2020-12-11:
I am in the same situation but I can provide controller details:

```
Controller:

  Product ID:	0x0b00
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	4.08
  Serial Number:	3032363330303234323137303039
  Speed:	Up to 12 Mb/s
  Manufacturer:	Microsoft
  Location ID:	0x14420000 / 4
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0
```

Here's one of the adapters I tried with:

```Anker USB-C Hub Device :

  Product ID:	0x8338
  Vendor ID:	0x291a
  Version:	0.01
  Serial Number:	0000000000000001
  Speed:	Up to 480 Mb/s
  Manufacturer:	Anker                  
  Location ID:	0x14450000 / 2
  Current Available (mA):	500
  Extra Operating Current (mA):	0
```

And here's another adapter I tried:

```
USB2.1 Hub:

  Product ID:	0x0610
  Vendor ID:	0x05e3  (Genesys Logic, Inc.)
  Version:	6.54
  Speed:	Up to 480 Mb/s
  Manufacturer:	GenesysLogic
  Location ID:	0x14300000 / 7
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

Controller:

  Product ID:	0x0b00
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	4.08
  Serial Number:	3032363330303234323137303039
  Speed:	Up to 12 Mb/s
  Manufacturer:	Microsoft
  Location ID:	0x14330000 / 8
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0
```