360Controller/360Controller #1118: Upgraded to Big Sur, now it doesn't work and cannot uninstall

ospfranco, 2020-11-13:
### Type of Controller
Replace this text with: Xbox 360

### OS Version
macOS Big Sur

### Driver Version
Version 1.0.0-alpha.6

### Connection Method
Wired

### Response of Controller
None

### Response in Preference Pane
None

### Detailed Explaination of Issue
Upgraded to macOS Big Sur, I got some warning on the first boot (not sure if it was because of this driver)

However now the xbox controller does not work and I cannot uninstall the extension, when I go to the system preferences pane, and click on the uninstall button, all I get is a error window, any suggestions how to manually uninstall and re-install?

ospfranco, 2020-11-13:
I downloaded the uninstall.command from the latest release, this is the output when it runs:

```
Index Refs Address            Size       Wired      Name (Version) UUID <Linked     1  144 0                  0          0          com.apple.kpi.bsd (20.1.0) 8    2   11 0                  0          0          com.apple.kpi.dsep (20.1.0)     3  182 0                  0          0          com.apple.kpi.iokit (20.1.0)    4    0 0                  0          0          com.apple.kpi.kasan (20.1.0)    5  185 0                  0          0          com.apple.kpi.libkern (20.1.    6  163 0                  0          0          com.apple.kpi.mach (20.1.0)     7   93 0                  0          0          com.apple.kpi.private (20.1.    8  107 0                  0          0          com.apple.kpi.unsupported (2    9    2 0xffffff80031de000 0xe000     0xe000     com.apple.kec.Libm (1) FA774   10   12 0xffffff80033fc000 0x8d000    0x8d000    com.apple.kec.corecrypto (1.   11    0 0xffffff8003490000 0x8000     0x8000     com.apple.kec.pthread (1) 37   12    1 0xffffff8003552000 0x2000     0x2000     com.apple.driver.watchdog (1   13   42 0xffffff80026d2000 0x2000     0x2000     com.apple.iokit.IOACPIFamily   14   50 0xffffff8002c17000 0x28000    0x28000    com.apple.iokit.IOPCIFamily    15   14 0xffffff8001890000 0x19000    0x19000    com.apple.driver.AppleSMC (3   16    2 0xffffff800128e000 0x77000    0x77000    com.apple.driver.AppleACPIPl   17    5 0xffffff8002d2f000 0x2000     0x2000     com.apple.driver.IOSlaveProc   18    3 0xffffff8001876000 0x10000    0x10000    com.apple.driver.AppleSEPMan   19    1 0xffffff8001587000 0x4000     0x4000     com.apple.driver.AppleBusPow   20   14 0xffffff80019dc000 0x4000     0x4000     com.apple.driver.usb.AppleUS   21    3 0xffffff8003044000 0x1000     0x1000     com.apple.driver.AppleUSBHos   22   23 0xffffff8002f49000 0x8d000    0x8d000    com.apple.iokit.IOUSBHostFam   23    3 0xffffff80031c8000 0xf000     0xf000     com.apple.driver.KernelRelay   24    1 0xffffff80015fb000 0x50000    0x50000    com.apple.driver.AppleCreden   25    3 0xffffff8001668000 0x6000     0x6000     com.apple.driver.AppleEfface   26    0 0xffffff8001672000 0x6000     0x6000     com.apple.driver.AppleFDEKey   27    2 0xffffff8001b78000 0x7000     0x7000     com.apple.kext.CoreTrust (1)   28    9 0xffffff80017fc000 0x15000    0x15000    com.apple.driver.AppleMobile   29    4 0xffffff8002c52000 0x14000    0x14000    com.apple.iokit.IOSCSIArchit   30   11 0xffffff8002d37000 0x12000    0x12000    com.apple.iokit.IOStorageFam   31    1 0xffffff8002c77000 0xe000     0xe000     com.apple.iokit.IOSCSIBlockC   32    1 0xffffff80031a4000 0x1a000    0x1a000    com.apple.iokit.IOUSBMassStorageDriver (184.40.6) 1D1E24DB-C7AC-390C-9B73-C9B139D3355F <30 29 22 14 7 6 5 3    33    2 0xffffff80018e9000 0x8000     0x8000     com.apple.driver.AppleUSBTDM   34    0 0xffffff8001757000 0x80000    0x80000    com.apple.driver.AppleKeyStore (2) 2D1ED2DB-F2C2-3530-B30C-9EC4B0220160 <33 28 27 24 23 18 17 10 8 7 6 5 3 1   35    0 0xffffff80018b1000 0x7000     0x7000     com.apple.driver.AppleSSE (1   36    4 0xffffff8001aa3000 0x7000     0x7000     com.apple.iokit.CoreAnalytic   37    2 0xffffff80017f8000 0x3000     0x3000     com.apple.kext.AppleMatch (1   38    3 0xffffff8003210000 0x44000    0x44000    com.apple.security.sandbox (   39    2 0xffffff8003207000 0x8000     0x8000     com.apple.security.quarantin   40    0 0xffffff8001b80000 0x4c000    0x4c000    com.apple.iokit.EndpointSecu   41    1 0xffffff8002850000 0x6000     0x6000     com.apple.iokit.IOBluetoothP   42   11 0xffffff8002c4d000 0x3000     0x3000     com.apple.iokit.IOReportFami   43    4 0xffffff8002764000 0x9c000    0x9c000    com.apple.iokit.IOBluetoothF   44    0 0xffffff800288d000 0xc000     0xc000     com.apple.driver.DiskImages    45    0 0xffffff800295a000 0x1000     0x1000     com.apple.iokit.IOKitRegistr   46   13 0xffffff800298e000 0x17000    0x17000    com.apple.iokit.IONetworking   47    3 0xffffff8002e7e000 0x1b000    0x1b000    com.apple.iokit.IOTimeSyncFa   48    0 0xffffff8001284000 0x9000     0x9000     com.apple.nke.applicationfir   49    1 0xffffff80016ca000 0x10000    0x10000    com.apple.security.AppleImag   50    0 0xffffff8001905000 0xe000     0xe000     com.apple.AppleSystemPolicy    51    0 0xffffff800132b000 0x2000     0x2000     com.apple.driver.AppleAPIC (   52   11 0xffffff80028ca000 0x5d000    0x5d000    com.apple.iokit.IOHIDFamily    53    3 0xffffff8002cd0000 0x1000     0x1000     com.apple.iokit.IOSMBusFamil   54    0 0xffffff8001318000 0x5000     0x5000     com.apple.driver.AppleACPIEC   55    0 0xffffff800188c000 0x3000     0x3000     com.apple.driver.AppleSMBIOS   56    0 0xffffff80018ac000 0x4000     0x4000     com.apple.driver.AppleSMCRTC   57    0 0xffffff8001313000 0x2000     0x2000     com.apple.driver.AppleACPIBu   58    1 0xffffff8001652000 0x2000     0x2000     com.apple.driver.AppleEFIRun   59    2 0xffffff8001655000 0xa000     0xa000     com.apple.driver.AppleEFINVR   60    0 0xffffff80018ba000 0x10000    0x10000    com.apple.driver.AppleSmartB   61    0 0xffffff80031db000 0x1000     0x1000     com.apple.private.KextAudit    62    1 0xffffff80030e7000 0x4e000    0x4e000    com.apple.driver.usb.AppleUS   63    0 0xffffff800313e000 0x28000    0x28000    com.apple.driver.usb.AppleUS   64    0 0xffffff8002ea6000 0x63000    0x63000    com.apple.iokit.IOUSBFamily    65    0 0xffffff8003046000 0x2000     0 [... cut ...]  2 0xffffff8002cbe000 0xb000     0xb000     com.apple.iokit.IOSCSIParall  191    0 0xffffff7fa1650000 0xb000     0xb000     com.highpoint-tech.kext.High  192    0 0xffffff7fa16f1000 0x2f000    0x2f000    com.softraid.driver.SoftRAID  193    0 0xffffff7fa165f000 0x85000    0x85000    com.highpoint-tech.kext.HighUninstalled 360Controller successfully.5-9996CA7A54B9 <190 14 6 5 3>
```

However I still cannot uninstall the extension from the system preferences and I cannot install the latest version

Sn00perton, 2020-11-13:
Also having problem of Xbox One controller not recognized after Big Sur update. Is there an alternative?

watchoutalien, 2020-11-13:
same issue here

Clashbuster, 2020-11-13:
same here, updated to big sur today no longer working.

c0d3x, 2020-11-13:
Yup same issue her when using the installer.
[image]

Sn00perton, 2020-11-14:
So I fixed this issue for myself but it is not what you want to hear haha.

To uninstall the 360 Controller Pref Pane I used CleanMyMacX (program I already had) and under the Extensions tab the old 360 Controller can be removed 

To get my Xbox One controller to work with my game again I downloaded and installed Parallels and Windows, bought on discount through my post secondary software store. Paired Xbox One controller to Windows, run Steam and game through Windows, and now everything works fine for my game.

KingWASD, 2020-11-14:
This is due to 360Controller being a "legacy" system extension. The 360Controller preferences pane eventually just disappeared for me some time after failing to uninstall.

rdonadono, 2020-11-14:
same issue

raffaeleg4, 2020-11-14:
Same. I was aware, though. There was a warning starting with macOS Catalina, that legacy extension wouldn't work anymore with the next version of macOS. I still have the Preference Pane, but with no reaction. Drew Mills developer is already allowed, but nothing works. 
[image]

thimoti26, 2020-11-14:
hi, I have the same problem after upgrading to Big Sur :'(.
I have the problem above (allow driver, without seeing it in security) . 
I tried to reinstall the app (apha.6), nothing more. 
I have this on startup if it could help : 
[image]

davidprati, 2020-11-15:
Same issue

amoneptah, 2020-11-16:
Well, I guess we all in here for the same reason... If anyone find a way (which not include downloading a Windows partition on my Mac, haha...) let us know <3 !

ospfranco, 2020-11-16:
Please, stop commenting without any additional information, it only spams the maintainers/owners of the repo, you can like the original issue and that should be enough

jpv123, 2020-11-21:
It looks like this is a problem related with the kext of the package since the Big Sur update. A few console logs show the following information which could be useful:

`library rebuild request failed: Extension with identifiers com.mice.driver.Xbox360Controller,com.mice.driver.WirelessOneController,com.mice.driver.Wireless360Controller,com.mice.driver.WirelessGamingReceiver not approved to load. Please approve using System Preferences.`

`Kernel Extension BLOCKED: Kext (G947KKM5RL, com.mice.driver.Xbox360Controller) version: b6d2d39ae77703c5c2c3f995d68a539252e515b2`

@FranticRain @ospfranco Can you please take a look?

FranticRain, 2020-11-21:
360Controller doesn't work in Big Sur and never will. This project is dead. Use a Bluetooth Xbox One controller if you can, since they are now officially supported. Same with the Dualshock 4.

ospfranco, 2020-11-23:
Alright, you heard the man, nothing to see here, closing this issue

closed 2020-11-23 by ospfranco

silentj, 2020-12-30:
I modified the steps from [this discussion](https://developer.apple.com/forums/thread/651132?answerId=650654022#650654022) to get the driver working for my wired Microsoft Xbox 360 controller on Big Sur 11.1. 

No guarantee this will work for the other controller types and it's likely the driver will break with a future release of macOS. But if you don't have any other controller options, this might help.

* Reboot to recovery mode (CMD+R at boot) and open up the terminal
* Disable System Integrity Protection: `csrutil disable`
* Reboot
* Install the [1.0.0-alpha.6 release](https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.6)
  * I still saw "The installation failed" when installing.
* Allow the driver in System Preferences -> Security & Privacy
  * If you don't see the prompt to allow a driver by "Drew Mills", then something has gone wrong and the above steps haven't worked.
* Reboot to recovery mode again and open up the terminal
* Add an exemption for the 360 drivers: `spctl kext-consent add G947KKM5RL` 
  * I got the ID of the driver using [this method](https://technology.siprep.org/getting-the-team-id-of-kernel-extensions-in-macos-10-13-and-higher/)
* Re-enable System Integrity Protection: `csrutil enable`
* Reboot

njrmay, 2021-02-12:
> I modified the steps from [this discussion](https://developer.apple.com/forums/thread/651132?answerId=650654022#650654022) to get the driver working for my wired Microsoft Xbox 360 controller on Big [...]

omg thank you this worked for me

NagoLazaro, 2021-02-25:
> I modified the steps from [this discussion](https://developer.apple.com/forums/thread/651132?answerId=650654022#650654022) to get the driver working for my wired Microsoft Xbox 360 controller on Big [...]

@FranticRain @ospfranco The solution shared by @silentj  worked for me right away. I'm Big Sur 11.1.1 and my Xbox 360 wireless controller works perfectly.

michael80-8008, 2021-05-21:
> I modified the steps from [this discussion](https://developer.apple.com/forums/thread/651132?answerId=650654022#650654022) to get the driver working for my wired Microsoft Xbox 360 controller on Big [...]

Works for me too