360Controller/360Controller #1119: Xbox X Series Controller not connecting

jabib0, 2020-11-14:
### Type of Controller
Xbox X Series Cotroller

### OS Version
macOS Catalina - Version 10.15.7

### Driver Version
360Controller 1.0.0-alpha.6

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Vibrate when plugged in via USB C

### Detailed Explanation of Issue
When I access the controller via System Preferences, I see "No Devices Found"

hpapagaj, 2020-12-14:
Is this model 1914?

matejsemancik, 2020-12-14:
Hi, I can confirm this with model 1914. My environment and behaviour is exactly the same as described in the issue.

Vendor ID: `0x045e  (Microsoft Corporation)`
Product ID: `0x0b12`

willem-aart, 2020-12-30:
Any updates on this? Just got a Series X Controller. Almost everything works via Bluetooth, except for the triggers. Would love to get it up and running using this driver :-)

GuillaumeRx, 2020-12-30:
@willem-aart after updating the controller with a windows machine I was able to use the controller via bluetooth natively. This driver should be used if you intend to play with the controller wired - MacOS supports natively xbox Series X controller via bluetooth

AnthonySpault, 2020-12-30:
> after updating the controller with a windows machine

Or a xbox ;)

gistgist, 2020-12-30:
I tried to update via the windows Xbox accessory app but the app kept crashing. Any other work around? No access to an Xbox.

willem-aart, 2020-12-30:
> @willem-aart after updating the controller with a windows machine I was able to use the controller via bluetooth natively. This driver should be used if you intend to play with the controller wired - MacOS supports natively xbox Series X controller via bluetooth

Do the triggers (LT & RT) also work for you on macOS after updating?

GuillaumeRx, 2021-01-01:
> Do the triggers (LT & RT) also work for you on macOS after updating?

They work fine for me at least :)

willem-aart, 2021-01-01:
> > Do the triggers (LT & RT) also work for you on macOS after updating?
> 
> They work fine for me at least :)

Hmm that's interesting. Updated my controller's firmware yesterday. When I try to assign the triggers in DiRT Rally, nothing happens. Other buttons work fine. Any ideas?

GuillaumeRx, 2021-01-02:
> Hmm that's interesting. Updated my controller's firmware yesterday. When I try to assign the triggers in DiRT Rally, nothing happens. Other buttons work fine. Any ideas?

@willem-aart did you tried another game ?

willem-aart, 2021-01-03:
> > Hmm that's interesting. Updated my controller's firmware yesterday. When I try to assign the triggers in DiRT Rally, nothing happens. Other buttons work fine. Any ideas?
> 
> @willem-aart did you tried another game ?

Not yet. But I did try my friends Xbox One controller, which worked perfectly. Very strange that the Series X controller does not work for me.

L0VEFIRE, 2021-01-24:
The newer Xbox Controllers most likey will not work starting from the Elite 2, Xbox Series Controllers. FRom what I have read they all use differnt byte packets and have some other internal differences. The 360Controller drive doesn’t have support for these from my understanding. 

Big Sur is supposed to have native bluetooth support for both (but wireless only)
Steam gamepad Driver (appears when your controller is connected to the computer and you enter big picture mode) - this driver should also support the new controllers. 

But mind you, on Big Sur, you will need to manually approve the drivers for both. *since my elite 2 now works with steam i can say there is progress*