360Controller/360Controller #1121: Controller is not detected after reinstalling driver. "No devices found showing"

Kwijiboz, 2020-11-15:
### Type of Controller
Easy SMX ESM-9100 (Xbox360 type controller)
### OS Version
macOS Catalina 10.15.7
### Driver Version
0.16.11
### Connection Method
Wired

### Device Name and Info
>Device Name: Easy SMX ESM-9100
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Guide button lights up
### Response in Preference Pane
The controller doesn't respond
### Detailed Explaination of Issue
I installed the driver to use with two Easy SMX controllers and it worked flawlessly, but after experiencing some problems trying to play steam games with both controllers connected I decided to reinstall the driver. So I uninstalled it and reinstalled it. After that, both controllers were no longer detected by the driver, although they both light up when connected and they both show in the USB hardware section.
During my second installation of the driver I noticed I didn't get any Privacy and Security messages telling me to allow the driver to be installed or anything, and I seem to recall that the first time I installed it I was asked to.
I've tried many solutions but no luck.
Thanks for your attention.

nnti3n, 2020-11-15:
Yea me too, I'm on 10.15.7

Kwijiboz, 2020-11-16:
I installed the latest alpha version and it recognized the controllers again, everything works fine now.

Dutprojet, 2020-11-18:
@Kwijiboz How ?

Kwijiboz, 2020-11-18:
> @Kwijiboz How ?

I uninstalled the version I was using which had stopped working (0.16.11) and installed the latest one here on hithub, the Alpha 6 version if I'm not mistaken. After that the controllers worked fine again

Kwijiboz, 2020-11-18:
The latest version is the 1.0.0-alpha.6