360Controller/360Controller #1123: I installed update for Big Sur - now the controller no longer found

ayahner, 2020-11-16:
I installed update for Big Sur - now the controller no longer found
### Type of Controller
Replace this text with:Xbox One

### OS Version
Big Sur - OSX 11.x

### Driver Version
latest (360Controller 1.0.0-alpha.6)

### Connection Method
Wired

### Device Name and Info
>Device Name:Apple T2 Controller
<br/>Product ID:0x8233
<br/>Vendor ID:0x05ac (Apple Inc.)

Divice not recognized - not listed in the preferences pane

mkb81, 2020-11-17:
Same problem with USB connection and macOS 11.0.1

Hapa333, 2020-11-17:
I have the same problem. I had latest (360Controller 1.0.0-alpha.6) before I updated to Big Sur but my controller was not showing up as a device so I uninstalled and tried reinstalling but it now (360Controller 1.0.0-alpha.6) is not installing as well giving a error.

Xrew, 2020-11-19:
I have same problem. After updating to Big Sur, the controller is not visible. I tried to reinstall driver, but installation failed.

cem6a, 2020-11-20:
Also having the same issue with the same controller, using the same version of macOS, on the same version of 350Controller.

FranticRain, 2020-11-22:
360Controller does not and will not work with Big Sur.

closed 2020-11-22 by FranticRain

k-funk, 2020-12-29:
> 360Controller does not and will not work with Big Sur.

@FranticRain might be good to put that at the top of the README if the project has reached its end of life.

FranticRain, 2020-12-29:
I am no longer in a position to be able to contribute to this repository (even README changes). The other maintainers have been informed and I know some of them have expressed interest in the continued existence of this repository. They have chosen to not take any action at this time. If you think that is a good idea, I would recommend tagging the other maintainers.

k-funk, 2020-12-29:
@MaddTheSane  @RodrigoCard

MaddTheSane, 2020-12-29:
Read-me updated.