360Controller/360Controller #1124: The installation failed.

t4thh, 2020-11-16:
### Type of Controller
Xbox One

### OS Version
Big Sur

### Driver Version
Newest Version 16/11/2020

### Connection Method
N/A

### Detailed Explaination of Issue
Everytime I come near finishing installation an error message says 'The installer encountered an error that caused the installation to fail. contact the software manufacturer for assistance'

Thanks

hachirotahoshino, 2020-11-19:
Hi, same here:

tried installing latest b6

**Type of Controller**

Xbox 360 Wheel

**OS Version**

Big Sur 11.01

**Driver Version**

Newest Version 16/11/2020

**Connection Method**

USB

**Detailed Explaination of Issue**

Everytime I come near finishing installation an error message says 'The installer encountered an error that caused the installation to fail. contact the software manufacturer for assistance'

[image]

here is the installer log:

Nov 19 16:33:11 ANEMONE Installer[11185]: Package Authoring Error: <background_scaling> has an unsupported MIME type: X-NSObject/NSNumber
Nov 19 16:33:11 ANEMONE Installer[11185]: Package Authoring Error: <background_alignment> has an unsupported MIME type: X-NSObject/NSNumber
Nov 19 16:33:11 ANEMONE Installer[11185]: Package Authoring Error: <layout-direction> has an unsupported MIME type: X-NSObject/NSNumber
Nov 19 16:33:11 ANEMONE Installer[11185]: Failed to load specified background image
Nov 19 16:33:19 ANEMONE installd[1446]: PackageKit: ----- Begin install -----
Nov 19 16:33:20 ANEMONE installd[1446]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "Une erreur s’est produite pendant l’exécution des scripts du paquet « Install360Controller.pkg »." UserInfo={NSFilePath=./postinstall, NSURL=file:///Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg, PKInstallPackageIdentifier=com.mice.pkg.Xbox360Controller, NSLocalizedDescription=Une erreur s’est produite pendant l’exécution des scripts du paquet « Install360Controller.pkg ».} {
	    NSFilePath = "./postinstall";
	    NSLocalizedDescription = "Une erreur s\U2019est produite pendant l\U2019ex\U00e9cution des scripts du paquet \U00ab\U00a0Install360Controller.pkg\U00a0\U00bb.";
	    NSURL = "file:///Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg";
	    PKInstallPackageIdentifier = "com.mice.pkg.Xbox360Controller";
	}
Nov 19 16:33:21 ANEMONE Installer[11185]: Install failed: Le programme d’installation a rencontré une erreur qui a provoqué l’échec de l’installation. Contactez l’éditeur du logiciel pour obtenir de l’aide.
Nov 19 16:33:21 ANEMONE Installer[11185]: Displaying 'Install Failed' UI.
Nov 19 16:33:21 ANEMONE Installer[11185]: 'Install Failed' UI displayed message:'Le programme d’installation a rencontré une erreur qui a provoqué l’échec de l’installation. Contactez l’éditeur du logiciel pour obtenir de l’aide.'.

Thanks

Jayroo5245, 2020-11-20:
Same Issue as above ^^

FranticRain, 2020-11-22:
360Controller does not and will not work with Big Sur.

closed 2020-11-22 by FranticRain

serveapps, 2021-12-03:
Is there a technical reason? If not, maybe we can raise some funds to get it working or pitch in code.