360Controller/360Controller #1125: Xbox One controller does not show up in preferences window but does show up in system report

SynWav, 2020-11-17:
### Type of Controller
Microsoft Xbox One connected with USB-C to USB-3

### OS Version
MacOs High Sierra 10.13.6

### Driver Version
360Controller 1.0.0-alpha.6, also have tried quite a few previous versions

### Connection Method
USB (Wireless controller plugged in with USB cable)

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 0x0b12
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Vibrates upon plugging in but does not light up or show up in preferences window of 360 driver.  

### Response in Preference Pane
No response.  "No devices found"

### Detailed Explaination of Issue
Controller vibrates upon plugging in but does not light up and does not show up in preferences window.  It does show up in system report under "controller" so I am certain I am using a data cable. I click "enable" driver" when attempting to connect the controller and it doesn't make any difference.  I have tried a few previous versions of the driver with no success.  The first time installing this driver I clicked enable on the "drew mills" message under the general tab of security and preferences.  However, I have installed and uninstalled the driver many times since then and the tab to click enable has not come back, even when I check for it immediately after installing.  Thank you.

obichidi, 2020-11-18:
I am experiencing the same issues.. the  green x light flashes but no input.

hpapagaj, 2020-12-14:
I have same problem. This is a Series X controller, model 1914. It has bluetooth too, but it doesn't recognised either. On gamepad-tester it is recognized when I connect trough bluetooth, but it doesn't work system-wide.