360Controller/360Controller #1126: Can't uninstall (macOS Catalina)

julie-impuls, 2020-11-22:
### OS Version
macOS Catalina 10.15.7

### Driver Version
Version 1.0.0-alpha.3

### Detailed Explaination of Issue
If I click the 'uninstall' button under the 'advanced' tab, I get a popup to confirm. When I click on 'yes' and enter my password, I get a new popup with the following message: 'Error Uninstalling the Driver!'.
Is there any other way to uninstall?

timtaylor4, 2020-11-22:
Hi, I'm having the same problem. I've tried to install the latest driver and am getting a 'Failed' message - I then tried to uninstall the one I have on system preferences and getting the error message.
The driver was working perfectly until I upgraded to Big Sur.

erniep24, 2020-11-23:
im having the same issue after updating to big sur, any fixers?

subirats345, 2020-11-24:
Same to my. I can't install on Big Sur

FranticRain, 2020-11-24:
The latest releases include an `uninstall.command`. If you download that and double click it, it should manually uninstall the driver without the errors from the app. If that doesn't work, you can open up that file in your favorite text editor and see exactly what files you need to delete manually.

closed 2020-11-24 by FranticRain