360Controller/360Controller #1128: CANNOT ENABLE DRIVER OR DETECT CONTROLLER BigSur 11.0.1

kckx91, 2020-11-29:
### Type of Controller
Xbox 360

### OS Version
BigSur 11.0.1
 
### Driver Version
1.0.0 - alpha 6

### Connection Method
Wired

### Device Name and Info
N/A

### Detailed Explaination of Issue
When attempting install, the install fails after it appears to complete.

The controller settings show up in System Preferences, but if I try to enable driver, it freezes, and never enables or detects a controller.

[image]

vonjur, 2020-11-29:
I have this same problem

Campbell1892, 2020-11-29:
Me too, tried repeatedly

gretel, 2020-11-29:
same same 😼

ponzius, 2020-11-30:
Same problem here.

diarmaidlindsay, 2020-11-30:
The developer has been working on this issue since it was first reported in March : 
https://github.com/360Controller/360Controller/issues/1027
You gotta go back to Catalina for now if you want to retain support for "Legacy System Extensions" since Apple decided to remove support for them in Big Sur and this developer has to work hard to rewrite it using Apple's new SDK...

givliano, 2020-11-30:
@diarmaidlindsay will the 360 controller work with the new SDK rewrite?

gretel, 2020-11-30:
@diarmaidlindsay this is not perfectly consistent with reports of people getting it to work on big sur.

Agastya-Asthana, 2020-12-01:
Is there any temporary solution to this or do I have to wait till the developers figure this out?

plexus77, 2020-12-01:
Is @FranticRain still working on a fix for this, his profile says he is not available to work on open source projects.

FranticRain, 2020-12-01:
As stated in similar issues that have since been closed, 360Controller will not work in Big Sur. Kernel extensions are deprecated in Big Sur, which is what 360Controller is. Conceivably a replacement could be written in DriverKit, however I believe that users should transition over to Bluetooth versions of Xbox controllers, as they have official support from Apple. This official support means a petter experience for both users and game developers. And I definitely don’t want to be responsible for fragmenting that ecosystem.

On top of that, as stated previously, I am no longer able to contribute to open source, so I won’t be making any further changes to this repository.

blaadje, 2020-12-01:
Sad, would love to have usb 360Controller on big sur 😢

cem6a, 2020-12-07:
That's very sad because some of my games only work when I use 360Controller and won't work with a wireless controller.

S1M0NP93, 2020-12-08:
Okay so I have been able to get my 'Xbox One' wireless controller working. I had an xbox 360 wired controller but like many of you it was not being detected. So 'for me' I downloaded 'Joystick Mapper', I had Enjoyable and ControllerMate but they did not work. Also you need to delete xbox 360 controller driver. Please do not try previous versions as it will slow down you mac. (Not I am using an iMac).

Once you have installed Joystick Mapper, go to the settings of the game you are planning to play & take the time to work out the keyboard setup then map your controls in joystick mapper to match your keyboard. 

IMPORTANT
Make sure you check the settings of your game. For me, when playing Borderlands 3 my controller was doubling up on buttons as I had the gamepad setting turned 'on'. 

Hope this helps till we get a proper driver. Also if this helps you spread the word. As I am yet to see anyone with a fix. 
Happy Gaming!

L0VEFIRE, 2020-12-17:
So first off this Driver currently doesn't have real support for Big Sur do to the changes added in this version of the macOS, while there are some tricky ways to get it to work it takes patience since its more of a hit or miss. also it might help if you use this unreleased version below with Elite 2 support or some of you. (not promised to work but I have had luck with it) you might have to install and uninstall the drive a few times before it will work. (also!! sometimes you need to connect your controller to windows via a cable and the xbox accessories app and so it syncs the intervals of the controller (idk why but this seems to always work for me) then boot into macOS. you always have to use a cable with Mac and xbox controllers unless you have a wireless adapter (Bluetooth will not work unless the game has official supported "apples gamepad framework") just use the cable. (if you already paired you controlled to bluetooth with Mac, you just put a lockout on the controller from using cable connection, so to reset it back to cable you have to remove the connection completely from macOS bluetooth settings and then either use bootcamp windows or a window pc to reset the controller to use a cable. make sure the controller is always turned off before switching OS's. its gets finicky. 

NOTE: You'll know the driver is functioning when the test mode registers your controller presses, BUT even when you make this far you make only have partial success since only some games will work. since macOS Big Sur requires games to use "apples game controller framework" which supports xbox one controllers and newer. I hope this helps some of you.

https://github.com/360Controller/360Controller/files/3844619/360ControllerInstall.dmg.zip

cem6a, 2020-12-17:
@L0VEFIRE I'm not able to get that release to install.  I've tried it a few times.  I get "The installation failed.  The Installer encountered an error that caused the installation to fail.  Contact the software manufacturer for assistance."

Thanks for the information and help, though.  I'll keep waiting.  Borderlands 3 only works when my XBOX One controller is connected via cable and using 360Controller as the driver.  I've submitted a bug report, and they acknowledged the bug, but I doubt they really care to fix it (it's been a bug since the release of the game).  Another person in a different thread used button mapping, and I tried that.  But, some of the controls are impossible to map correctly, and I'm too frustrated to keep working at it -especially after the hours I've already spent troubleshooting.

Anyways, thanks again.

L0VEFIRE, 2020-12-18:
@cem6a So I uninstall the driver after updating to Big Sur, just now, I was using a beta version and it worked (but it will always say it failed, but in some cases it will work) now I was not able to get that to work this time with either build (posted and alpha 6) Just know the the system preference have had serious bugs in it for a long time and apple has never fixed it no matter how many time I reported them. so ehhh that's apple for you. anyway. 

my best suggestion to you depending how much space you have on your Mac, is to do why I have gave in to do. Install Bootcamp windows say 200GB since it conserves about 10% of that and play your gams on there, and make sure to get "MacFanControl" App and have it always set to Full Blast. then you good to go. I play games like Sekiro and Tomb Raider just fine with some tweaks to game settings for performance reasons. and best of all bc its windows full xbox controller support! yes! 

might be worth it for you! (as long as you already have a copy of either windows 7, 8, key or 10, they all work with Win 10) 

hope this helps!! best work around. (trying to get xbox controllers to work on Mac twith games that are not Apple Arcade is just going to be a nightmare, until companies make the move to the new ecosystem apple has created with Big Sur) 

need any help let me know.

cem6a, 2020-12-18:
@L0VEFIRE Thanks for the info.  You're probably right; Bootcamp with Windows is probably the best option, albeit not the easiest.  I haven't used Windows in like 10 years, and I would have to free up a lot of space for the installation.  I'm not really stressing that I can't play my game, as I'm not a huge gamer anyway.  But, it's frustrating when I spent a lot of money on the game(s) and controller and can't use them.  I'm trying to be patient.

I was able to complete the 0.16.11 installation, but it doesn't show in the sys pref pane.  I also received a notification of a macOS update, which I JUST completed (not sure if that matters, but just an FYI), which makes me on macOS 11.1.

I'm still hopeful that there's a way to fix this without Windows.  Since macOS doesn't support the 360 controllers natively, I can't just buy an older controller.  With BL3, I have to tick the 360 controller emulation even though I'm using my One controller.  I'll just keep waiting for a fix or until I get too frustrated and start the Bootcamp process.

Thanks, again.

L0VEFIRE, 2020-12-18:
@cem6a 

Welcome, personally you have a 2 year stretch for the new generation of macOS and Tech to take place, and we both know Catalina works fine with both. Might be worth it to either downgrade to Catalina or like before do Bootcamp. I know how frustrating tech can be so im glad I can help guide you in whatever way possible. (Catalina should still be on app store, but it could be complicated trying to go backwards since Big Sur uses a new firmware hence the return of the boot sound) maybe if you do go that route install a second partition with Catalina just to play your game with you controller, until you've beaten it. just a thought. but at least now you have some options. :D 

Best of luck

Ahmadre, 2020-12-29:
> As stated in similar issues that have since been closed, 360Controller will not work in Big Sur. Kernel extensions are deprecated in Big Sur, which is what 360Controller is. Conceivably a replacement could be written in DriverKit, however I believe that users should transition over to Bluetooth versions of Xbox controllers, as they have official support from Apple. This official support means a better experience for both users and game developers. And I definitely don’t want to be responsible for fragmenting that ecosystem.
> 
> On top of that, as stated previously, I am no longer able to contribute to open source, so I won’t be making any further changes to this repository.

Thank you soooo much for your answer :)! I used my Bluetooth-Controller with this Driver over USB 😆 , but now finally I can use my XBox One Controller over Bluetooth. Thanks Apple ❤️

L0VEFIRE, 2021-01-10:
I found someone who clevery resolved the controller issue on Big Sur for Steam, while its not directly related to this it has some upsides worth looking into. It more how to get software/drivers to be approved by big sur's sys. pref. security. which we all know blocks it. check it out. 

[Fix for game controllers on Big Sur!!!  Steam for Mac.pdf](https://github.com/360Controller/360Controller/files/5792049/Fix.for.game.controllers.on.Big.Sur.Steam.for.Mac.pdf)

cem6a, 2021-01-10:
@L0VEFIRE Very clever!  Thanks for that info.  Maybe I'll try to look up the Developer Team ID for 360Controller and give it a try in terminal using recovery mode.

FranticRain, 2021-01-10:
My team ID is G947KKM5RL, while it lasts.

FatSandwich, 2021-01-10:
Did it work? I have the Original Xbox One controller which doesn't have Bluetooth so I need this. Also, it shows 360Controllers in settings.

L0VEFIRE, 2021-01-12:
@FatSandwich 

yes, this does restore wired connection, i have tested and confirmed this for myself. 

solution is here: (#1 on possible solutions)
https://github.com/360Controller/360Controller/issues/1139

sophiebean13, 2024-02-23:
> I have this same problem

has this 

> Thank you soooo much for your answer :)! I used my Bluetooth-Controller with this Driver over USB 😆 , but now finally I can use my XBox One Controller over Bluetooth. Thanks Apple ❤️

I'm using the same controller the driver is intended for. I don't know much about coding. Do you have to know at least some coding to download this? If not, the driver isn't working as of today Feb23, 2024. I'm still new to this kind of thing for gaming. Is there any other driver you can suggest to download? I am using a macbook pro 2019 and have the latest update. I'm trying to play sims but Sims is only letting me play through Origin and not through steam. The driver won't work at all and I'm not sure if I've even installed it correctly. I went through the process but it hasn't even detected it. Please and thanks to you or anyone else that sees and responds to this!