360Controller/360Controller #1130: Not working on macOS Big Sur

Powne, 2020-12-05:
### Type of Controller
3rd party pdp wired Xbox One

### OS Version
macOS Big Sur

### Driver Version
360Controller 1.0.0-alpha.6

### Connection Method
Wired

### Device Name and Info
>Device Name: Wired controller fo Xbox One
<br/>Product ID: NA
<br/>Vendor ID: pdp

### Response of Controller
No lights turn on, controller not detected and can't tik the "allow drivers" box

### Detailed Explaination of Issue
Hi! This driver is great and I appreciate all the work you've put in. It worked fine for me with OS Catalina, today I updated to Big Sur and my controller wasn't detected. I opened the control scheme and the "enable driver" box was unmarked, I tried to tik it, but it would't let me (I have these drivers allowed in System preferences). So i uninstalled, rebooted, and dowloaded the driver again, but it won't let me install them. it fails in the installation and tells me to contact the software developer. (Sorry for my broken English, I am Italian)

Hope you can help me fix this!

jdavid82, 2020-12-06:
I have the same problem, however my computer which is running Big Sur 11.0.1 can still detect the controller through `Apple Menu -> About this Mac -> System Report -> Overview -> Hardware -> USB` and it shows it as `PDP Wired controller for Xbox One`. 

[In this section](https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb) the author mentions that `At this time, PDP and PowerA controllers are unsupported by this driver as of macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code.`.

However, I believe that the sentence (and the paragraph it's contained in) was written in the context of `the controller not being detected or the usb cable not working` and not in the context of `the installer doesn't work`.

As you can see at the beginning of my comment, I mentioned that my controller was being detected by the OS.

I like these controllers, they are widely available in many shopping centres in my city, so it would be a shame if they no longer supported Mac OS as Big Sur gets rolled out (considering how Mac OS and iOS have such a large percentage of their user base upgrading to the latest version in comparison with users of other operating systems).

Thanks

Guivems, 2020-12-10:
Hi everyone,

same issue for me.

Hope your guys will find a solution.
Thanks :)

matszafraniec, 2020-12-12:
Hello,

same issue here.

Type of controller:
Microsoft Xbox One controller

OS Version:
macOS BigSur 11.0.1

Driver version:
360Controller 1.0.0-alpha.6

Connection method:
wired

Detailed Explaination of Issue:
Same as explained in topic. System won't let install the driver. It fails during the the installation and tells me to contact the software developer.

aleksbatista, 2020-12-14:
Type of controller:
Microsoft Xbox 360 Controller

OS Version:
macOS BigSur 11.0.1

Driver version:
360Controller 1.0.0-alpha.6

Connection method:
Wired

Detailed Explaination of Issue:
Same boat as above comments. Wondering if I should just get a bt controller instead as this seems to be a difficult issue to work around.

rahshin, 2020-12-16:
Type of controller:
Microsoft Xbox 360 Controller

OS Version:
macOS BigSur 11.0.1

Driver version:
360Controller 1.0.0-alpha.6

Connection method:
Wired

Detailed Explaination of Issue:
Same boat as above comments. 
Here is the screenshot
[image]

Neogene, 2020-12-19:
Sadly same problem.

hassankhan, 2020-12-20:
Dupe of #1128. Answered [here](https://github.com/360Controller/360Controller/issues/1128#issuecomment-736584705).

jdavid82, 2020-12-21:
@hassankhan what's the answer?

victorwads, 2020-12-22:
Same here, not installing

dugobu, 2020-12-23:
same here, error at installation, however it is present in my system preferences but doesn't recognize any xbox one joystick whether it's usb-c or wireless.

gregcsw, 2020-12-26:
same here, error at installation, and now present in system in system preferences. Tried using the uninstall button under the advanced tab and it won't uninstall just say there was an error.

glezmen, 2021-01-02:
same here
Wired XBOX 360 controller
Is listed in System report/hardware/USB
After rebooting the Mac, I can tick the 'enable' box, but it says apps from Drew Mills needs to be enabled (the usual dialog), but in SystemPreferences it is not even mentioned :-/

hachirotahoshino, 2021-01-08:
Hi, dunno if it'll help, but i managed to get  the wired xbox360 kext to load by following this guide:

https://steamcommunity.com/discussions/forum/2/3004429475624592660/

and replacing the dev id with the one of this software:

G947KKM5RL

My xbox 360 wheel is now recognized in ETS2 in steam

MaFtuk, 2021-01-18:
Same issue here with macOS BigSur 11.0.1 - however the setting shows up in the System Preferences. I cannot, however, uninstall it as that gives an error...

Is there a way to manually uninstall?