360Controller/360Controller #1133: No device found, no response. XBox One Wired connection. MacOS Big Sur M1 processor

commodrone, 2020-12-20:
### Type of Controller
Xbox One

### OS Version
Version 11.1
### Driver Version
Version 1.0.0-alpha.6

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x02dd
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
No response, not founded

### Response in Preference Pane
No response, not founded

### Detailed Explaination of Issue
No response, not founded. It appears on the USB Device Tree but that's all, no response everywhere else. I am writing this just to make sure this is a fixable thing since my MacMini as the new M1 processor. Maybe that is what is causing the issue. Ty guys, first time im running into problems.

hasanbasri1993, 2020-12-25:
yes, same with me after upgrade to bigSure No device found

Zlitus, 2020-12-26:
Same issue here on my MacBook Pro M1. I also had an error during the installation of the PKG, Preferences System -> Security&Privacy didn't proposed me to allow the developer 😢

dreamingthink, 2020-12-27:
Hello,

I have tried to compile the project with Xcode. It is working if you build only XboxOneBluetooth
and change macos 10.11 to macos & also change "Sign to run locally" (I don't have developper account)

https://mega.nz/file/I49wxJgA#_Tx_vd2Gr3XqW6XesJjOv2jDs96SFe5C4Q2xsn3bWVQ
You have to put the 3 files in /Library/Extensions and execute the command :
sudo chown -R root:wheel XBOBTFF.plugin.dSYM
sudo chown -R root:wheel XBOBTFF.plugin
sudo chown -R root:wheel XboxOneBluetooth.kext

Now i'm able to see my controller in https://gamepad-tester.com/ with Bluetooth.
Unfortunatly, it is not working on Stadia for the moment.
Any idea ?

GamesOfSummer, 2020-12-27:
Same issue, would prefer an in-house fix ASAP.

dreamingthink, 2020-12-28:
Hello, I'm not sure you have seen the answer of the developper of 360Controller : https://github.com/360Controller/360Controller/issues/1128#issuecomment-736584705