360Controller/360Controller #1136: Xbox adaptive controller not detected on Big Sur !

thibs6x, 2020-12-28:
### Type de contrôleur Manette adaptative Xbox  ### Version du système d'exploitation macOS 11. 1  ### Version du pilote 360controller 1.0.0-alpha.6  ### Méthode de connexion Câble USB  ### Nom et informations sur l'appareil https://www.xbox.com/fr-FR/accessories/controllers/xbox-adaptive-controller  ### Réponse du contrôleur Au démarrage de l'iMac la manette clignote et finit par s'éteindre car elle n'est pas prise en charge par big sur malgré le pilote mais j'ai essayé de faire tourner shadowPC et la manette s'allume mais ne fonctionne toujours pas ni en macos ni en shadowpc ! Et quand j'éteint shadowpc la manette reste allumée mais ne fonctionne pas  ### Explication détaillée de la question Impossible de détecter le joystick par 360controller même avec le joystick allumé et impossible d'activer le pilote ! Je suis vraiment désespérée car ce matériel m'aide vraiment dans mon quotidien car je suis handicapée et j'utilise mon ordinateur à l'aide de mon nez et de ma bouche ! Désolé pour mon anglais J'utilise la traduction Google parce que je parle français

[image]

uotw, 2020-12-29:
Confirmed same issue on Big Sur here. Not sure if it makes any difference, but with Apple silicon M1 MBA.

thibs6x, 2020-12-29:
@uotw Same but with an old model

[image]

Ahmadre, 2020-12-29:
On my mac I can activate it, but this message pops up:
[image]

thibs6x, 2020-12-29:
@uotw Try to unlock the privacy and security settings before installing 360controller that fixes the problem of this pop-up

Ahmadre, 2020-12-31:
Solved by connecting over Bluetooth:)

thibs6x, 2021-01-02:
@Ahmadre Thanks it works!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! ;)

thibs6x, 2021-01-02:
But I'm still happy to find a solution to stay connected via usb because after several minutes of inactivity the controller disconnects

Ahmadre, 2021-01-02:
> But I'm still happy to find a solution to stay connected via usb because after several minutes of inactivity the controller disconnects

Same, but my Controller starts to disbehave and force holding the down direction.

Apple has to do this better.