360Controller/360Controller #1137: Expired Certificate Problem

KellerChad, 2020-12-31:
### Type of Controller
Xbox One

### OS Version
macOS Big Sur 11.1

### Driver Version
Can't Install

### Connection Method
Bluetooth

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
Can't Install

### Detailed Explanation of Issue
When I go to install 360Controller 1.0.0-alpha.6 my computer will not finish the install fails saying it encountered an error. When I checked the read me, I looked in my security preferences, and at no point did it prompt me to trust the certificate. I tried clicking the lock icon in the upper right-hand corner of the install window and it says the certificate expired Thursday, December 17, 2020, at 12:30:27 AM Eastern Standard Time. I think this may be the problem I am having. Also, there is a drop-down box to select trust but it keeps resetting and not allowing me to trust the certificate since it is expired.

JujuAdams, 2021-01-01:
Having the same issue here.

NicBas48, 2021-01-09:
Having the same issue.

L0VEFIRE, 2021-01-10:
check out my new post might help you