360Controller/360Controller #1138: No devices found while attempting to connect a wireless xbox one s controller via micro usb.

LE-E-dev, 2021-01-07:
### Type of Controller
Xbox One(s)

### OS Version
mac os BigSur 11.0.1

### Driver Version
the alpha 2 driver

### Connection Method
via micro usb(witch is a data transfer cable)

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: idk just the controller that came with the xbox, late 2015 imac 27inch 5K retina
<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
light turns on, also vibrates.

### Response in Preference Pane
totally nothing

### Detailed Explaination of Issue
"https://gamepad-tester.com/" also works but it only says that there are no devices found in the system prefrences tab. pls help!!

LE-E-dev, 2021-01-07:
also light turns on only when I open https://gamepad-tester.com/

MaciekBaron, 2021-01-07:
The driver does not work on Big Sur and the original developer sadly does not have time to update it right now.

LE-E-dev, 2021-01-07:
> The driver does not work on Big Sur and the original developer sadly does not have time to update it right now.

Thnaks! Didn't know it.

closed 2021-01-07 by LE-E-dev