360Controller/360Controller #1139: ⚠️ READ BEFORE POSTING - BigSur & 360Controller

L0VEFIRE, 2021-01-10:
I enjoy helping others, especailly with this issue that effects us all, but first thing to address is what the owner and developer has stated about the current state and future of this driver; it's reached the end of the road come Big Sur. Here is the words qouted by FranticRain himself to help everyone clearly understand and help keep redundant posts.

**FranticRain:**
_> As stated in similar issues that have since been closed, 360Controller will not work in Big Sur. Kernel extensions are deprecated in Big Sur, which is what 360Controller is. Conceivably a replacement could be written in DriverKit, however I believe that users should transition over to Bluetooth versions of Xbox controllers, as they have official support from Apple. This official support means a better experience for both users and game developers. And I definitely don’t want to be responsible for fragmenting that ecosystem.
> 
> On top of that, as stated previously, I am no longer able to contribute to open source, so I won’t be making any further changes to this repository._

(2) the second issue is simple, the Apple M1 and any future M Processors have absolutely no support for this driver as far as I have heard and from my understanding unless  you use an Intel Mac it wont work for you.

(3) this is where we come to some partial resolve of making the controllers work on Big Sur, no promises here but so far it has worked for most ppl from what i have read on it. I have posted the forum i snapshoted as PDF, read it if you want to explore a resolution to getting installed programs/drivers accepted by apples system preferences. 

[Fix.for.game.controllers.on.Big.Sur.Steam.for.Mac.pdf](https://github.com/360Controller/360Controller/files/5792061/Fix.for.game.controllers.on.Big.Sur.Steam.for.Mac.pdf)

FatSandwich, 2021-01-10:
Thanks, so this works on M1 Big Sur??

L0VEFIRE, 2021-01-10:
> Thanks, so this works on M1 Big Sur??

no, as far as I’m concerned it wont work at all. (there is someone who complied there own personal build of it for the M1, but without it being signed your out of luck)

Johnth10, 2021-01-10:
Xbox series controller works after firmware update on windows or on the xbox

L0VEFIRE, 2021-01-10:
@Johnth10 
could you be more specific for everyone. (1) which mac processor are you referring to and presume you also mean Big Sur. (2) are you referring to wired or wireless connection

Johnth10, 2021-01-10:
1 : M1
2 : both

Read this : https://github.com/360Controller/360Controller/issues/1119

L0VEFIRE, 2021-01-12:
@Johnth10 

thanks for clearing that up, this is good to know! I will add this to the main post once I clean up the markdown formatting a bit to make it easier to read. I also update my xbox one firmware and can confirm even on Intel macs it works for both bluetooth and wired with no issues. 👍

Johnth10, 2021-01-12:
@L0VEFIRE 

Good to know. Thank you for your work.

FatSandwich, 2021-01-15:
I tried my Xbox controller on Parallels and it worked with a wire which is kind of odd. But now it doesn't work.

L0VEFIRE, 2021-01-15:
> I tried my Xbox controller on Parallels and it worked with a wire which is kind of odd. But now it doesn't work.

I have had nothing but nightmares in past attempts to use virtualization software to play games; which is why its not a listed option. yeah, it might have worked for a moment but this is something ive found to not go away. which is why boot camp is a more stable option. this issue it’s beyond my scope, so if your determined maybe you can google for some answers. only hint i can tell you is to shut down the emulation and check out your settings for Usb/accessories. best i can tell ya.

FatSandwich, 2021-01-15:
The problem is that I'm on the M1 Mac and it runs ARM processors so it doesn't work. Parallels have a beta which makes Windows 10 work but it could run into issues maybe that's why. But, if Bootcamp comes with support I'd switch to it. Thanks!

L0VEFIRE, 2021-01-15:
oh right forgot about that. well someone was able to make bootcamp work on the M1. they did something with the arm build for windows microsoft has. um i saved it i will look it up for you maybe it might help you.

L0VEFIRE, 2021-01-15:
> The problem is that I'm on the M1 Mac and it runs ARM processors so it doesn't work. Parallels have a beta which makes Windows 10 work but it could run into issues maybe that's why. But, if Bootcamp comes with support I'd switch to it. Thanks!

okay here are two links that might help.

[1st person to run windows on M1] 
https://9to5mac.com/2020/12/02/windows-on-m1-macs-run-arm-virtualization/

[using parallels to run windows on M1]
https://www.google.com/amp/s/9to5mac.com/2020/12/22/run-windows-10-arm-m1-mac-apple-silicon-x64-x86-no-boot-camp-video/amp/

FatSandwich, 2021-01-15:
Thanks, I'm currently using Parallels now.

bw881, 2021-01-17:
**@L0VEFIRE** 
Many thanks, followed instructions as posted. Now can use xbox controller freely on big sur! 
Mac model: Macbook pro mid2014.

FatSandwich, 2021-01-17:
How?

L0VEFIRE, 2021-01-17:
@FatSandwich 

he is using a Intel Processor hence the mac model he is using.

L0VEFIRE, 2021-01-17:
@FatSandwich 

this might be worth looking into if you haven’t read it already, but i would read it very carefully, a lot of important steps that must be followed for it to work correctly.

https://kb.parallels.com/en/124771

FatSandwich, 2021-01-18:
Aren’t those steps for BlueTooth though? My controller doesn’t have BlueTooth. Thanks anyway man!

L0VEFIRE, 2021-01-18:
@FatSandwich 
i might actully have a hunch on why your wired controller isnt working, and its uniqueto VM’s, but you also have a M1 so the software is quite new for M1. But in normal instances when using a wired device with VM’s it will installed the driver on first conncetion, but once you go to reboot the VM the driver is still installed, but treats it like it never was so you end up in limbo. 

what version of xbox controller are you using by chance.? full name and model (should be on back)

FatSandwich, 2021-01-18:
Model: 1537
Name: Xbox One Controller
Bluetooth: No

Also, what's weird is that in the device's settings in Windows 10 it shows "Controller - Connected". But, when I go to https://gamepad-tester.com/ it says "No Gamepad Detected". Another weird thing is when I first connected my Xbox controller to my M1 mac it worked flawlessly in Parallels and I could even use it to play games, but now it just doesn't work!

Thanks for your help!

FatSandwich, 2021-01-18:
I have a similar issue to
https://answers.microsoft.com/en-us/windows/forum/all/driver-error-for-xbox-controller-on-windows-10/1d7b20f5-dde7-44f4-b94b-ca3fdaebad8f

But they didn't seem to fix the issue.
I still haven't tried reinstalling windows but if nothing works I might have to!

L0VEFIRE, 2021-01-18:
reference this link 
https://www.google.com/amp/s/www.drivereasy.com/knowledge/solved-microsoft-xbox-one-controller-driver-install-failed-issue/amp/

go to method 1 uninstalling the driver. see if that helps. 

but i will say windows 10 on its own has issues with pairing xbox controllers. yours is wired so maybe a different cable,.

see if uninstalling helps if so might be on to something

FatSandwich, 2021-01-18:
I tried uninstalling the driver but that didn't work either. At this point, I'll try reinstalling Windows 10.

FatSandwich, 2021-01-19:
IT WORKED!!! REINSTALLING WINDOWS 10 WORKED!!! I should have tried it before! Thank you so much for your help man!

FatSandwich, 2021-01-19:
And now it's not working again!!!!! It did the same thing as last time!!!

L0VEFIRE, 2021-01-19:
@FatSandwich

This is interesting, that removing the driver did nothing yet re-installing the OS made it work again. It must have cleared something that was present, that is getting stuck or something when you connect your control. I wonder if this is a M1 thing. 

I would strongly suggest you join the conversation on parellels fourms and also start a support ticket to get his looked into.

Here is the Fourm for M1

https://forum.parallels.com/forums/parallels-desktop-16-for-m1-mac-technical-preview.786/

Here is the main support page.

https://www.parallels.com/support/

I really wish i had an M1, so i could look into this more. I hope in time someone can answer you question once you post your issues and get help.

FatSandwich, 2021-01-19:
Yeah I hope so! Thank you I will go the forums and ask them!

FatSandwich, 2021-01-19:
Also, I realized that when I first connected it showed Controller - Connected with a controller icon. But now it shows a computer and phone.

[image]
[image]

FatSandwich, 2021-01-19:
Unfortunately, I tried using an app called "Driver Verifier" built into the computer. Then it kept on saying "your Windows Insider Build ran into a problem." I'm going to try to install Windows 10 again hence it takes about 10 minutes.

FatSandwich, 2021-01-19:
The weird thing about this issue is that nobody is having trouble with it?

FatSandwich, 2021-01-19:
This time when I reinstalled Windows 10 my controller didn't work! This is a very very weird issue!

L0VEFIRE, 2021-01-19:
@FatSandwich
Ah yes, all these errors i have come across myself using an intel mac, im going to bet my suspicions it part the fact VMs emulate hardware (this doesn't seem to be going well, never did for me when i tried so many  times), the last part is Windows 10 has problems of its own. A shame really. This is why i personally never get a new type of processor before its saturated itself in the market and all the issues can be ironed out; thats just me. Anyway. All you can really do now is have patience since this isn't likely going to get fixed over night and might takes some months. As long as you reported it. I would keep an eye on the full list of all the issues in the Parrellels M1 fourm. Tons of issues in there. 

Also given your current statemate, might need to find a different means to play games.

FatSandwich, 2021-01-19:
Thanks so much for your help. I guess I'll just wait for a response. I can play on keyboard and mouse so it won't be that big of an issue.

KitsuneCobain, 2021-01-20:
Hey guys, I really miss my clone hero at the minute and came across this potential fix for my wired 360 xplorer to work again on my macbook. I've never really done anything like this on mac os before and am worried about any chance I could mistype something or mess something up? Is this something that someone with zero experience can do?

L0VEFIRE, 2021-01-20:
@KitsuneCobain 

hi, yes while it might seem advanced, anyone can do it. You just have to know how. In your case i can explain in more detail how to do this.

First thing your going to want to do is reboot you mac, the moment the screen goes black, hold these keys on the keyboard “Command + R” until the apple logo appears with a loading bar underneath. Then let go, once loaded you should be loaded into macOS recoevry. From here you might need to sign in if you have icloud active, but either way next is to go to the top right menubar and open the terminal

One the terminal is open this is were you just need to take your time, and double check your work, no 
___
**STEPS for installing xbox driver :**
1. Boot into Recovery mode (hold ⌘-R at boot). Open Terminal (Utilities -> Terminal).
2. Type ```spctl kext-consent add G947KKM5RL```
3. Press return. 
4. Type ```kmutil trigger-panic-medic --volume-root "/Volumes/Macintosh HD"```
 _(If driver name is different then replace “Macintosh HD” with whatever your drive is named; press Tab at the end of typing "/Volumes/" to get a hint)._
5. Press return. 
6. Reboot.
7. Once at desktop, load terminal
8.  Type ```sudo kmutil load -p /Library/Extensions/360Controller.kext```
---

While your still in the terminal on masOS recovery, repeat this process for Steam gamepad driver as well since its also needed if your using steam.

**STEPS:**
1. Boot into Recovery mode (hold ⌘-R at boot). Open Terminal (Utilities -> Terminal).
2. Type ```spctl kext-consent add MXGJJ98X76```
3. Press return. 
4. Type ```kmutil trigger-panic-medic --volume-root "/Volumes/Macintosh HD"```
 _(If driver name is different then replace “Macintosh HD” with whatever your drive is named; press Tab at the end of typing "/Volumes/" to get a hint)._
5. Press return. 
6. Reboot.
7. Once at desktop, load terminal
8.  Type ```sudo kmutil load -p /Library/Extensions/SteamInput.kext```
---

Then you can boot into macOS, install the xbox driver, then install the “big picture game pad driver” for steam,  ignore the failed message plug in an xbox one controller and it works. The only step that might casue and issue is step 8, but that is optional.

KitsuneCobain, 2021-01-20:
> @KitsuneCobain
>
> hi, yes while it might seem advanced, anyone can do it. You just have to know how. In your case i can explain in more detail how to do this.
>
> First thing your going to want to do is [...]

Thank you so freaking much man, I'm very happy to be able to play again, bless :)

P3tras, 2021-01-21:
Thank you very much! Got it working with Steam. I am was trying to connect 2 controllers to Steam. Xbox 360 wired and Stadia controller (also via wired connection). Stadia controller was working fine without doing any changes, but xbox controller was not detected. Long story short, by following your guide, I got Xbox 360 controller working even without installing Steam drivers. However, now Stadia controller was not detected no matter what I did (installed Steam Big picture controller drivers and all other steps). What helped me is installing newest Steam Beta version. Not sure if it just reset generic controller drivers or there are some fixes. But might help someone who want to use different controllers.

L0VEFIRE, 2021-01-21:
@P3tras

Wired connections are different then wireless. the xbox driver is needed for wired connections, steam driver is used for both wireless and wired (without the steam driver one cant use bluetooth connections)

also while i have found in odd occasions that the xbox driver will sometimes work on big sur without that terminal method it wont be stable and at some point stop working. 

so glad this was able to help you. it might be worth noting that steam recently updated there gamepad driver last update publicly to partially support elite 2 controllers. so going beta was a clever idea.

FatSandwich, 2021-01-21:
Yeah, it also worked for me on Big Sur for a while but in Parallels which is odd.

FatSandwich, 2021-01-21:
Also, why is nobody working on creating drivers for Xbox One? Is it hard?

L0VEFIRE, 2021-01-21:
Anything is hard until one understands it. Big Sur is different is uses DriverKit, while all previous macOS versions used Kernel Extensions which is what this driver (360Controler) is a kernel extension; that has a control panel as a prefence pane. **Unless someone who is well versed in how to write in apples DriverKit and  wants to see this driver updated, it will never be. The Future os contrllers is wireless**; that being said the future of most tech is wireless. Wired conncetions are being depreciated. So Might be time to invest in a new controller. I did. I had a xbox one and had to get a one that was both wireless and wired. It works for me. 

plus someone already mentioned above they us M1 and xbox one and got theres to work fine after reading this post. So i think you real problem is relying on Parellels, over a more stable and proven moethod that will allow your conroller to work.

FatSandwich, 2021-01-21:
Ok, I guess it is time to buy a Bluetooth controller. But why did Apple have to change it?

L0VEFIRE, 2021-01-21:
Well first off make sure the controller is xbox or a supported controller of Big Sur and this driver (link in post).

L0VEFIRE, 2021-01-21:
To ansaeer your question about why apple made the change, there focus is to modernize the outdated. And using kernal space is problmatic. Think about anytime you used a driver or program and something when wrong with it, then everything crashed the mac shutdown or had a kernel panic. (That is bc that app or driver was made using kernel extensions (in kernal space), the reason for switchin to Driverkit is bc is relies on User Space meaning if tehre is an error or something goes wrong, it wont bring down the whole system with it essential speaking. Its more secure relibable.

FatSandwich, 2021-01-21:
O

L0VEFIRE, 2021-01-21:
The thing is there is a 2 year stretch not just for Apple Silicon, but for having all developers apply apples gamepad framework which allows (xbox one, Elite contrllers) to work wirelessly only. But that means developers have to learn swift and driverkit first that takes TIME.

L0VEFIRE, 2021-01-21:
Another reason apple made the switch to its own chips (CPU, GPU, ect) was bc there was to many manufacture flaws in there intel line up. Flaws in the Intel CPUS, GPUS. And other compentents. It made more sense for apple to just build a whole new system that made sense and they had control of; it hard to manufacture a product when you dealing with like 5 differnt companies you cant control. Lol

clark-b124, 2021-01-25:
Xbox 360 Xplorer guitar hero controller (wired)
Macbook Pro Big Sur

I don't have much coding experience myself and I'm thirsting for some Clone Hero. First of all I'm assuming I don't need the steam setup (I'm not familiar). On step 8 of the first part I get the error message "no such file or directory." Am I doing something wrong in the reboot part?

L0VEFIRE, 2021-01-25:
@clark-b124 
nope your good, that just means you haven’t installed the driver yet. step 8 only is for checking everything works, but you need to install the driver before doing it. also step 8 is optional.

ja0b, 2021-01-30:
Thanks, this worked for me I'm using Big Sur 11.1 (20C69). I'll add some keywords here so google searches point to this thread more often.

- macOS 11 Big Sur - Xbox One controller not detected and Steam games not detected
- Big Sur 11.1 Xbox One controller not detected 
- Xbox One controller not detected

dreamingthink, 2021-02-01:
Any news with 11.2 update ?

L0VEFIRE, 2021-02-01:
What about it?

Deep Gratitude,
Stefan

maybeageek, 2021-02-04:
Thanks for the good work everybody involved! I used this driver a lot and have fond memories of macOS gaming.

Just as a sad sidenote: I despise bluetooth for its input lag and jittery control, so I always preferred the wireless dongle that comes with the xbox controllers. Way more precision, way faster responses. So sorry to see this driver go the way of the dodo :-(

clark-b124, 2021-02-04:
Does it work with Big Sur 11.2? anyone tried?

FatSandwich, 2021-02-04:
I tried it in the browser and it worked. But now it doesn’t work,

onvt, 2021-02-04:
No problem on Big Sur 11.2 for me. 😃

clark-b124, 2021-02-04:
Still new to coding and trying to get this to work. I can't find the files to delete when manually uninstalling the driver, probably because I don't know where they'd be. I did command+shift+H in finder and no luck. Is there a way I can search for them?

L0VEFIRE, 2021-02-05:
yeah my bad, I forgot to mention, that in order to even find those files you need to have access to your drive, which apple has removed from being on by default. in order to turn them on you need to go to your desktop and make sure your Finder is active on the menubar, then click on "finder", then click on Preferences, then click on the "sidebar" tab at the top of that window. from here you want to go to the bottom section call "locations" its a bit hard to see the text. but then make sure your name's MacBook is checked. then when you open the finder window on your sidebar you will now see that option, from here you can get to the path I laid out. (I wish GitHub allowed photos would be easier to make my guide) 

Summary of steps:
(1) Click on the desktop or the finder on the dock
(2) click on "finder" in the menu bar
(3) click on preferences
(4) select the "sidebar" tab
(5) check the box for "your name"'s Macbook (whatever your devices is)
(6) exit this window and go to your finder window
(7) navigate to Macintosh HD > Library ...( so forth now follow instructions from guide)

clark-b124, 2021-02-05:
Thanks a ton. I couldn't find...

Macintosh HD/Library/Extensions/Wireless360Controller.kext
Macintosh HD/Library/Extensions/WirelessGamingReceiver.kext

...under Extensions, but trashed (and cleared trash) the rest.

L0VEFIRE, 2021-02-05:
Okay, tested under macOS 11.2 and can confirm it works just fine. (using Xbox One Controller, Model 1708)

[186 more replies]