360Controller/360Controller #1140: GPD WinMax Built-in Xbox 360 Controller failed with errors

kingo132, 2021-01-14:
### Type of Controller
GPD WinMax Built-in Controller

### OS Version
Catalina 10.15.7

### Driver Version
1.0.0-alpha.6

### Connection Method
Built-in, using USB interface, can be treated as USB wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller for Windows
<br/>Product ID: 0x12
<br/>Vendor ID: 0x2F24

### Response of Controller
Driver failed to start

### Response in Preference Pane
Driver failed to start

### Detailed Explaination of Issue
Driver failed to start with these logs
```
Xbox360Peripheral::probe(Controller for Windows)
Xbox360Peripheral::start(Controller for Windows) <1>
Unknown device release 100
start - failed to read chatpad setting (e000404f): 0xe000404f (UNDEFINED)
Xbox360ControllerClass:0x1000003af start

[ElementContainer] Failed to open report descriptor: 0xffffc988

[ElementContainer] Descriptor: ec14ff497df864d0726174696f6e5f31353933363433333930303531223a7b22656e61626c6547656e65726963223a66616c73652c22656e61626c65427942726f77736572223a7b226368726f6d65223a747275652c2265646765223a747275652c2266697265666f78223a747275652c22736166617269223a66616c73657d2c22656e61626c654279446f6d61696e223a5b7b22646f6d61696e223a2273616c6573666f7263652e636f6d222c226d6174636853756264ec14ff49

0x1000003af: parse report descriptor failed: 0xe00002bc
Xbox360ControllerClass:0x1000003af failed to parse report descriptor

Xbox360ControllerClass:0x1000003af start failed

Xbox360ControllerClass:0x1000003af stop
```

kingo132, 2021-01-14:
From the google result, this game controller is from: 
ShenZhen HuiJiaZhi Technology Co., Ltd.
深圳市慧嘉智科技有限公司
which vendor id is: 0x2F24
This company also made many other xbox controllers

L0VEFIRE, 2021-01-19:
Here is a list of all compatible controllers, as far as the dev is concerned from what I heard there is no more additions being done to this driver. Your best bet is using a controller that is already supported, also note since this driver is retired, only my workarounds will allow the drive to work and must follow them in order to get a successful wired connection. see my post "READ BEFORE POSTING" https://github.com/360Controller/360Controller/issues/1139 once you have a compatible controller. 

https://github.com/360Controller/360Controller/wiki/Compatible-controllers

b00t0x, 2021-01-23:
@kingo132 My GPD WIN Max controller is recognized as a Xbox controller so it works with this software by default.
- Product ID : 0x028e
- Vendor ID : 0x045e (Microsoft  :flushed: )

So imprements might differ by the revision of the GPD WIN Max.

Could you test this Hackintosh-style Info.plist injector with OpenCore ?
https://gist.github.com/b00t0x/0d8cd3d70523c21c885d771f82875252
(My GPD WIN Max doesn't need this so I can't test it.)

kingo132, 2021-01-23:
> @kingo132 My GPD WIN Max controller is recognized as a Xbox controller so it works with this software by default.
> 
> * Product ID : 0x028e
> * Vendor ID : 0x045e (Microsoft  😳 )
> 
> So imprements might differ by the revision of the GPD WIN Max.
> 
> Could you test this Hackintosh-style Info.plist injector with OpenCore ?
> https://gist.github.com/b00t0x/0d8cd3d70523c21c885d771f82875252
> (My GPD WIN Max doesn't need this so I can't test it.)

Thanks b00t0x, I replaced your info.plist to the project and compiled the 360controll.kext but the driver can't load. I don't know why.
I have added my custom device's product id and vendor id to the info.plist, so maybe it's not the info.plist problem. Maybe should modify the code, because it says "Failed to open report descriptor" and "parse report descriptor failed".

kingo132, 2021-04-02:
Add more information, the built-in joystick is successfully recognized by the driver, here you can see the screenshot below:
[image]
But the driver will output these error logs, I have added some logging code to print more information like return values.
And one thing need to mention, when I move the joystick or press the button, the driver do have responses and will print many logs like "Failed to send special message ..."
```
jason@mywm ~ % sudo dmesg | grep Xbox360Controller
Xbox360Controller: Unknown device release 100
Xbox360Controller: SendInit ret: e000404f
Xbox360Controller: SendInit ret: e000404f
Xbox360Controller: SendInit ret: e000404f
Xbox360Controller: start - failed to read chatpad setting (e000404f): 0xe000404f (UNDEFINED)
Xbox360ControllerClass:0x1000003df start
Xbox360ControllerClass:0x1000003df failed to parse report descriptor
Xbox360ControllerClass:0x1000003df start failed
Xbox360ControllerClass:0x1000003df stop
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
Xbox360Controller: read - failed to handle report: 0xe00002d7
[... 182 lines cut ...]
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
[... cut ...]
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
```

kingo132, 2021-04-03:
Got some information about the error code.
KernelIOKit/usb/USB.h | kIOUSBPipeStalled | 0xE000404F | Pipe has stalled, error needs to be cleared
KernelIOReturn.h | kIOReturnOffline | 0xE00002D7 | device offline

kingo132, 2021-04-03:
Digging deeper into the problem, I have found more logs in dmesg about this. I think the problem may be related to the error of getting report descripter.
```
Xbox360Peripheral::probe(Controller for Windows)
Xbox360Peripheral::start(Controller for Windows) <1>
Xbox360Controller: Unknown device release 100
Xbox360Controller: SendInit ret: e000404f
 CCFlags: 0x3, CCLevel: 5 ConsoleFlags: 0x0, ConsoleLevel: -1
 CCFlags: 0x4, CCLevel: 5 ConsoleFlags: 0x0, ConsoleLevel: -1
IO80211Controller::findAndAttachToFaultReporter no pipe found, making it
Xbox360Controller: SendInit ret: e000404f
Xbox360Controller: SendInit ret: e000404f
IOUSBHostDevice: family specific matching fails
IntelFirmware: Driver init()
Xbox360Controller: start - failed to read chatpad setting (e000404f): 0xe000404f (UNDEFINED)
IntelFirmware: Driver Probe()
IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
IO80211Controller::start _controller is 0xa966b61ad98b3abd, provider is 0xa966b61ad983c83d
IntelFirmware: Driver Start()
[HID] [MT] [Error] AppleMultitouchDevice::_cacheDeviceProperties Error: reportID 0x00 is 126 bytes long, 4 bytes remaining
PXSX::requestPowerDomainState
BSD root: disk1s5, major 1, minor 12
disk1s5: device is not readable.
hfs_mountfs: buf_meta_bread failed with 13
hfs_mount: hfs_mountfs returned error=13 for device unknown-dev
hfs_mountroot failed: 13
apfs_vfsop_mountroot:1502: apfs: mountroot called!
Xbox360ControllerClass:0x1000003d5 start

[ElementContainer] Failed to open report descriptor: 0xffffc988

[ElementContainer] Descriptor: 9853b31ca2c6caf2000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000d08c5eda15a5d95bb80a62a17fffffff010001000000000000000000000000000052e24080ffffff50a0eb4080ffffff000000009406000040e90000806a0300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000009853b31c

0x1000003d5: parse report descriptor failed: 0xe00002bc
Xbox360ControllerClass:0x1000003d5 failed to parse report descriptor

Xbox360ControllerClass:0x1000003d5 start failed

Xbox360ControllerClass:0x1000003d5 stop
```

kingo132, 2021-04-03:
I dumped the hid descriptor under windows, it seems very different from the failed descriptor in dmesg. So I'm wondering if the driver is reading the wrong descriptor
```
Information for device Controller for Windows (VID=0x2F24 PID=0x0016):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0009
Current configuration value: 0x01
Number of open pipes: 7

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0xFF	bDeviceClass      (Vendor specific)
0xFF	bDeviceSubClass   
0xFF	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x2F24	idVendor
0x0016	idProduct
0x0100	bcdDevice
0x01	iManufacturer   " "
0x02	iProduct        "Controller for Windows"
0x03	iSerialNumber   "77A6777F"
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0099	wTotalLength   (153 bytes)
0x04	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xA0	bmAttributes   (Bus-powered Device, Remote-Wakeup)
0xFA	bMaxPower      (500 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0x5D	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x00	iInterface

Unknown Descriptor:
------------------------------
0x11	bLength
0x21	bDescriptorType
Hex dump: 
0x11 0x21 0x00 0x01 0x01 0x25 0x81 0x14 0x00 0x00 
0x00 0x00 0x13 0x02 0x08 0x00 0x00 

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x04	bInterval         (4 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x08	bInterval         (8 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x01	bInterfaceNumber
0x00	bAlternateSetting
0x04	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0x5D	bInterfaceSubClass   
0x03	bInterfaceProtocol   
0x00	iInterface

Unknown Descriptor:
------------------------------
0x1B	bLength
0x21	bDescriptorType
Hex dump: 
0x1B 0x21 0x00 0x01 0x01 0x01 0x83 0x40 0x01 0x04 
0x20 0x16 0x85 0x00 0x00 0x00 0x00 0x00 0x00 0x16 
0x06 0x00 0x00 0x00 0x00 0x00 0x00 

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x83	bEndpointAddress  (IN endpoint 3)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x02	bInterval         (2 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x04	bEndpointAddress  (OUT endpoint 4)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x04	bInterval         (4 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x85	bEndpointAddress  (IN endpoint 5)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x40	bInterval         (64 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x06	bEndpointAddress  (OUT endpoint 6)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x10	bInterval         (16 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x02	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0x5D	bInterfaceSubClass   
0x02	bInterfaceProtocol   
0x00	iInterface

Unknown Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
Hex dump: 
0x09 0x21 0x00 0x01 0x01 0x22 0x87 0x07 0x00 

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x87	bEndpointAddress  (IN endpoint 7)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x10	bInterval         (16 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x03	bInterfaceNumber
0x00	bAlternateSetting
0x00	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0xFD	bInterfaceSubClass   
0x13	bInterfaceProtocol   
0x04	iInterface   "Xbox Security Method 3, Version 1.00, � 2005 Microsoft Corporation. All rights reserved."

Unknown Descriptor:
------------------------------
0x06	bLength
0x41	bDescriptorType
Hex dump: 
0x06 0x41 0x00 0x01 0x01 0x03 

Microsoft OS Descriptor:
------------------------------
0x12	bLength
0x03	bDescriptorType
Hex dump: 
0x12 0x03 0x4D 0x00 0x53 0x00 0x46 0x00 0x54 0x00 
0x31 0x00 0x30 0x00 0x30 0x00 0x90 0x00 

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  " "
0x02   0x0409  "Controller for Windows"
0x03   0x0409  "77A6777F"
0x04   0x0409  "Xbox Security Method 3, Version 1.00, � 2005 Microsoft Corporation. All rights reserved."

------------------------------

Connection path for device: 
?? USB xHCI ??????
Root Hub
Controller for Windows (VID=0x2F24 PID=0x0016) Port: 5

Running on: Windows 10 or greater (Build Version 19042)

Brought to you by TDD v2.17.0, Feb 23 2021, 14:04:02

```

kingo132, 2021-04-03:
I found the invalid descriptor is read from this property. I'm wondering which code set this invalid descriptor.
[image]

kingo132, 2021-04-04:
I think I found it, it was because the function Xbox360ControllerClass::newReportDescriptor failed to copy the descriptor. I don't know why it fails.
[image]
[image]

kingo132, 2021-04-04:
Finally got the report descriptor fixed by changing the option from kIODirectionOut to kIODirectionInOut, since the buffer->writeBytes requires the buffer has the kIODirectionIn option. Now the joystick works perfectly.
However, there are still SendInit and SendSwitch errors, and also many send special message errors.
```
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
Xbox360Controller: Failed to send special message 1f, ret: e000404f
Xbox360Controller: Failed to send special message 1e, ret: e000404f
```
[image]

Anyway, the joystick works now despite all these errors. And I will try to comment these log codes to use the joystick for a while to see if it will encounter any problems.

closed 2021-04-04 by kingo132