360Controller/360Controller #1141: MacBook goes to sleep when using the Xbox Controller

ironboy36, 2021-01-15:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

ironboy36, 2021-01-15:
I am getting angry when I am in the middle of playing a game with my Xbox controller my MacBook goes to sleep, is there any way to fix that?

L0VEFIRE, 2021-01-19:
Sounds familiar ... need information first though before I can determine if I can help or not. 

(1) What software are you using to play games (Steam / Apple App Store / Apple Arcade / Other)

(2) How are you using your controller (wired or wireless)

(3) Are you playing from Battery Power or Power Adapter

(4) what are you Mac power settings for the power source your using? _(sometimes the Mac doesn't recognize the app as a game and will treat it otherwise, since it doesn't consider a controller a means to keep the system active like a mouse cursor will.)_

L0VEFIRE, 2021-01-20:
Forgot a important question:

(5) what model and year mac do you have?