360Controller/360Controller #1142: Error Uninstalling the 360Controller

TacosAndPizza, 2021-01-19:
Controller: Xbox One

OS Version: macOS Big Sur Version 11.1

Driver Version: 360Controller 1.0.0-alpha.6

Connection Method: Wired

Device Name and Info:
>Device Name: XB1 Classic Controller
<br/>Product ID: 0x581a
<br/>Vendor ID: 0x24c6

Response of Controller: The guide button doesn't light up but it vibrates. All else is the same as any other controller, buttons and other controls work fine, as good as new.

Response in Preference Pane: The controller's reaction before the Big Sur update worked perfectly fine; all triggers, joysticks, and buttons reacted and appeared in the preference pane as expected.

Explanation of Issue:
My issue, is that I am unable to uninstall the 360Controller installation.

Backstory~ About a day ago, before I had read about the 360Controller no longer working with macOS Big Sur, I was trying to find the reason as to why my controller wasn't connecting with the program. I hadn't used my controller or tried connecting it since August of 2020, so I was unaware of the changes. Anyhow, I had thought to uninstall the program and reinstall it to see if that would change anything. With that said, I had uninstalled the 360Controller, and restarted my computer. Afterwards, I came back to GitHub, and selected the 360Controller 1.0.0-alpha.6 installation. Upon reinstalling it, after I had typed in my password to allow the installation, a pop-up message appeared saying that an error occurred and that I would need to contact the program manufactures for help. With that said, I tried restarting my computer and reinstalling it a couple more times just to make sure it wasn't my computer that was causing the issue at hand. Anyhow, after I was about done trying (as I had kept getting the same pop-up message) I opened my system preferences only to see that the program had actually installed. At that time I had once again tried to connect my controller (which I now know isn't possible with the macOS version I have at the moment), although after having no success, I thought I'd just uninstall the program after all.

All in all, when I go to uninstall the 360Controller, after typing in my password to allow the uninstallation, I am greeted with a message saying an error occurred with the uninstallation. 

[image]

Does anyone have an idea on how to fix this? I have tried restarting my computer, but no luck; I'm unsure of what other files or systems that the program may have installed into my computer, so I haven't tried looking for each file individually to delete, as I thought that might be a bad idea.

Thanks so much for reading, it's a small issue but wanted to explain exactly what had happened! x3
Have a great rest of your day!

L0VEFIRE, 2021-01-19:
Known issue, already posted a while ago. there is no fix, and will never be a fix. as stated by the dev.

The only work around is my post "READ BEFORE POSTING" they explains how to get around this. First you going to need to manually remove all the files ... yeah. then empty the trash and reboot into macOS recovery and go to the terminal. from here follow the steps I have listen in my post. you will then be able to use a xbox controller wired only, and for some games very few will let you use a xbox one controller via bluetooth.

TacosAndPizza, 2021-01-19:
@L0VEFIRE 

Thank you so much for your reply, I will follow your instructions and get things cleaned up on my computer. 

I apologize that it’s already a known issue; I was reading through the post “READ BEFORE POSTING”, but it was very late at night and some things just blew right over my head x3

I apologize, but thank you for taking the time to respond! I wish you well! :)

I will close the issue with this comment <3

closed 2021-01-19 by TacosAndPizza