360Controller/360Controller #1143: PDP xbox controller on MacOS big Sur 11.1

LordDasew, 2021-01-20:
Hello there,

I have macOS big sur 11.1 and it doesn't recognise my PDP xbox controller, it says to contact the developer as it's not updated the pkg... ...

Any help please?

LordDasew, 2021-01-21:
i forgot to say ... the message that appear when i try to install the driver ALPHA 6 is THE INSTALLATION FAILED, contact the software manufacturer...

LordDasew, 2021-01-22:
?

hugeugo, 2021-01-27:
Hi I am experiencing the same issue! It says the installation failed every time I install an alpha version, it seems to be installed though... 
But have the same problem, my PDP XboxOne controller doesn't appear either. I can see it in the About this mac > System Report > Hardware > USB and it also works perfecly when I use it under windows with bootcamp. 

Here is the information for the controller : 

PDP Wired Controller for Xbox One - Midnight Blue:

  Product ID:	0x02a0
  Vendor ID:	0x0e6f
  Version:	1.0f
  Serial Number:	000007DB58E19840
  Speed:	Up to 12 Mb/s
  Manufacturer:	Performance Designed Products
  Location ID:	0x14130000 / 2
  Current Available (mA):	500
  Extra Operating Current (mA):	0

I would be super grateful if you could add this controller to the supported ones! 
Thanks

hackftz, 2021-02-08:
same as I

Doublejay, 2021-02-22:
I have the same controller and it's not recognised by my m1 on Big Sur.

However, I'm curious to know how you 

> Hello there,
> 
> I have macOS big sur 11.1 and it doesn't recognise my PDP xbox controller, it says to contact the developer as it's not updated the pkg... ...
> 
> Any help please?

I have the same controller and it's not recognised by my m1 on Big Sur.

However, I'm curious to know how you go the driver installed ?  I get blocked by security.

LordMurillo, 2021-02-23:
Hello,
I tried to install the driver on Big Surf 11.2.1, but the installation fails.

The gamepad is detected correctly. About this mac > System Report > Hardware > USB:
Controller:
  ID del producto:	0x0b12
  ID del fabricante:	0x045e  (Microsoft Corporation)
  Versión:	5.01
  Número de serie:	3039373134373132343137303336
  Velocidad:	Hasta 12 Mb/s
  Fabricante:	Microsoft
  ID de la ubicación:	0x14110000 / 24
  Corriente disponible (mA):	500
  Corriente operativa adicional (mA):	0

alexfanshawebabylon, 2021-03-26:
duplicate of https://github.com/360Controller/360Controller/issues/1148