360Controller/360Controller #1144: BETOP Game handle 阿修罗se (Cable version) not work on macos big sur 11.2

hackftz, 2021-01-24:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

BETOP Game handle 阿修罗se (Cable version)

### OS Version
Replace this text with the version of macOS that you are running

macos big sur 11.2

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

the newest xbox360controller

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

cable (not wireless)

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

BETOP Game handle 阿修罗se (Cable version)

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

hackftz, 2021-01-26:
help help help!

alecyu, 2021-02-04:
tried this ?
https://github.com/360Controller/360Controller/issues/1139

hackftz, 2021-02-05:
> tried this ?
> #1139

thks, my friend, I will try it in this weekend.