360Controller/360Controller #1145: Cannot Unload Driver "In order to unload the driver, you need administrator privileges."

ianhu92, 2021-02-06:
### Type of Controller
Xbox 360

### OS Version
macOS big sur 11.2

### Driver Version
1.0.0-alpha6

### Connection Method
wired

### Device Name and Info
N/A

### Detailed Explaination of Issue
Trying to uncheck "Enable Driver", it asks for administrator access. 

But I have no idea how to run this configuration or the "System Preference" in sudo mode.

[image]

hribernik, 2021-02-12:
same problem here

cabaalexander, 2021-02-12:
Yep, I cannot uninstall it either

flamingfungus, 2021-02-21:
same happened with me, but then my everything went offline

L0VEFIRE, 2021-02-25:
This is a know issues, and the work around is in my post “READ BEFORE POSTING” you have to manually uninstall the driver. but first you have to manually add the driver to be approved in macOS recovery.