360Controller/360Controller #1146: Third-party controller not found but already added

brunaw, 2021-02-16:
## Type of Controller
Third-party Xbox 360

## OS Version
High Sierra 10.13.6

## Driver Version
1.0.0-alpha.6

## Connection Method
Wired

## Device Name and Info
Device Name: USB Gamepad (1 extra space before, 10 after)

Product ID: 0x0001

## Vendor ID: 0x0810 (Personal Communication Systems, Inc.)

## Response of Controller
Guide button lights toggle on press.

Response in Preference Pane
Controller does not appear on the preference Pane

## Detailed Explaination of Issue
The controller isn't recognised by the driver. This has previously been addressed in #1056 and solved with a new release, but that still does not work for me. The controller responds on https://html5gamepad.com/, with: "Connected: Yes", "Mapping: n/a"

clayvernon, 2021-02-21:
I came here for this same issue. 

I am frustrated that something that  [#1056](https://github.com/360Controller/360Controller/issues/1056) says it has been solved, but there are clearly people buying this pad and it's not working.

I am on Mojave. 10.14.6

brunaw, 2021-02-21:
@clayvernon I'm managing to use the controller by using a Windows 10 virtual machine called Parallels Desktop and using Steam through it, since Windows does not require any additional drivers to recognize the controller. It's a "solution" but very much imperfect because Parallels uses a good bit of space (around ~15Gb here) and it's paid after the 2-week trial period.  Maybe this can help you while we wait for a solution for the mac driver.

clayvernon, 2021-02-21:
@brunaw I'm guessing since @FranticRain has left the project, we are going to have to pay one way or the other.  
It's unfortunate that the info in #1056 was not correct.
Buying different gamepads is an option, and I suppose I can sell these at near the price I paid.