360Controller/360Controller #1148: Driver Installation Failed on Big Sur

djhalnon, 2021-02-27:
### Type of Controller
N/A

### OS Version
Mac OS Big Sur 11.2.2

### Driver Version
1.0.0-alpha.6

### Connection Method
N/A

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
N/A

### Detailed Explaination of Issue
Installation of driver fails. The error message is "The installation failed. The installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance." This error message does not disclose the actual nature of the error that caused the installer to fail.

[image]

Zara4397, 2021-03-06:
Got the same result, but got also a pop-up from Mac stating there are instances that needed to be updated. Probably just a matter of supporting Big Sur.

paulina222, 2021-03-09:
> Got the same result, but got also a pop-up from Mac stating there are instances that needed to be updated. Probably just a matter of supporting Big Sur.

the same thing happened to me.. Zara4397, did you ever find another way to use your xbox controller on your mac by any chance??

JamesDeCarlo, 2021-03-10:
Same Issue with Big Sur Version 11.2.2

romulobastos, 2021-03-20:
Same issue with Big Sur V 11.2.3

alfanse, 2021-03-22:
Same failure, console logs:
```
2021-03-22 17:47:50+00 my-MacBook-Pro package_script_service[1709]: PackageKit: Executing script "postinstall" in /tmp/PKInstallSandbox.zCtfsq/Scripts/com.mice.pkg.Xbox360Controller.VtNyyT
2021-03-22 17:47:50+00 my-MacBook-Pro package_script_service[1709]: Set responsibility to pid: 1825, responsible_path: /System/Library/CoreServices/Installer.app/Contents/MacOS/Installer

<hmmmm>
2021-03-22 17:47:51+00 my-MacBook-Pro package_script_service[1709]: ./postinstall: Major version is not 10. Cowardly failing out of script execution.
</hmmmm>

2021-03-22 17:47:51+00 my-MacBook-Pro package_script_service[1709]: Responsibility set back to self.
2021-03-22 17:47:51+00 my-MacBook-Pro installd[1242]: PackageKit: releasing Spotlight indexing
2021-03-22 17:47:51+00 my-MacBook-Pro install_monitor[1828]: Re-included: /Applications, /Library, /System, /bin, /private, /sbin, /usr
2021-03-22 17:47:51+00 my-MacBook-Pro installd[1242]: PackageKit: releasing backupd
2021-03-22 17:47:51+00 my-MacBook-Pro installd[1242]: PackageKit: allow user idle system sleep
2021-03-22 17:47:51+00 my-MacBook-Pro installd[1242]: PackageKit: Cleared responsibility for install from 1825.
2021-03-22 17:47:51+00 my-MacBook-Pro installd[1242]: PackageKit: Cleared permissions on Installer.app
2021-03-22 17:47:51+00 my-MacBook-Pro installd[1242]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "An error occurred while running scripts from the package “Install360Controller.pkg”." UserInfo={NSFilePath=./postinstall, NSURL=file:///Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg, PKInstallPackageIdentifier=com.mice.pkg.Xbox360Controller, NSLocalizedDescription=An error occurred while running scripts from the package “Install360Controller.pkg”.} {
	    NSFilePath = "./postinstall";
	    NSLocalizedDescription = "An error occurred while running scripts from the package \U201cInstall360Controller.pkg\U201d.";
	    NSURL = "file:///Volumes/360ControllerInstall/Install360Controller.pkg#Install360Controller.pkg";
	    PKInstallPackageIdentifier = "com.mice.pkg.Xbox360Controller";
	}
```

L0VEFIRE, 2021-03-30:
its going to fail it wasn't made for to work Big Sur the app is dead on Big Sur, you have to manually approve the driver but that is kinda pointless now since it won't work even after you do.
you need to start using bluetooth controller supported by apple. Xbox one or newer / PS4 or newer. unless you go back to Catalina and use this driver. 

if you using steam, then you need to load big picture modem and download the gamepad driver there. anywhere else should just work wirelessly.

djhalnon, 2021-03-31:
Interesting. What's behind the decision to prevent people from using wired controllers on Macs? It's one thing not to sell them, but another to actively prevent people who already have them, from ever using one. I honestly don't get it.

JamesDeCarlo, 2021-03-31:
Definitely disappointing. Used this for years on my mac before upgrading to big sur. I have newer xbox one controllers with bluetooth that work. But this driver also worked excellent for my logitech 710 controller. Which I miss. Thank you for the response though.

L0VEFIRE, 2021-03-31:
> Interesting. What's behind the decision to prevent people from using wired controllers on Macs? It's one thing not to sell them, but another to actively prevent people who already have them, from ever using one. I honestly don't get it.

Honestly, i asked apple many times in tge last year with no response, i even send product feedback, i think more ppl need to do in order to let them know how important it is to ppl. though it is very disappointing i agree, im forced to use boot camp to play games now.

iSashaCH, 2021-04-06:
Got my xbox one controller to work wired with macOS 11.2.3. Earlier I was struggling to "allow" driver from Drew Mills because
I there was simply no prompt in Security settings to Allow it. 

Inspired by https://steamcommunity.com/discussions/forum/2/3004429475624592660/  I went and figured out signers Team ID: G947KKM5RL

so, booted into Recovery Mode and executed:
`spctl kext-consent add G947KKM5RL`

followed by:
`kmutil trigger-panic-medic --volume-root "/Volumes/**{{replace with your macOS drive name}}**"`

upon reboot in Panic Medic mode I've been asked to allow all the third-party extensions by developers. Drew Mills was not one of them. After allowing those macOS prompted to reboot but I chose "Not Now" option and instead went to XBOX 360 Controller Pref panel - Advanced Tab and clicked "Enable Driver" (it was unchecked), wait second or two and get notification that I need to approve software from Drew Mills in Security settings - went to Security Settings and hurray - this time there was a prompt to do so! Allowed, rebooted and it works!
[image]
[image]

L0VEFIRE, 2021-04-06:
@iSashaCH 
yeah nice, not sure how you got your controller working. i already have a big post that explains this here. sorry you had to do it the longer way. still curious what made yours work though.

diego-lipinski-de-castro, 2021-04-23:
I followed @iSashaCH instructions and also got it to work on Big Sur

[image]

L0VEFIRE, 2021-04-23:
I'm glad you got it to work but I have a whole post on this that was made prior to this, why are we creating a new duplicate thread. 🤦 

Link:
https://github.com/360Controller/360Controller/issues/1139

yori21, 2021-06-18:
@iSashaCH's approach seems to **not** work on M1 Macs :(

L0VEFIRE, 2021-06-18:
> @iSashaCH's approach seems to **not** work on M1 Macs :(

As of currently, you need to use Windows on a M1 mac to get controllers to work. (This can be done via VM Software like Parrells or VMware … ect) 
I’m finally ordering a M1 mac tomm so when i get it i can do some more investigation work.

ruyrybeyro, 2021-10-10:
Following these steps,   the last version of the extension  appears grayed out in the Security setting of Big Sure. It let´s itsef be selected, but does not appear loaded at boot. Any comments?

letalvoj, 2021-12-04:
Solution which worked for me: https://github.com/360Controller/360Controller/issues/1180#issuecomment-986002973

First install `0.16.11`, make sure you get to the point where the settings appear in the system, yet it says no device. Then install `1.0.0-alpha6`.

alvarusan, 2024-02-04:
Solution that worked for me on 12.6.8 (Monterey) [#1148 comment](https://github.com/360Controller/360Controller/issues/1148#issuecomment-813735388)

Darkdonald2, 2024-02-17:
Disabling system integrity protection (SIP) then trying to enable the driver again worked for me on macOS Sonoma.
A search on google will tell you how to disable SIP  
Unfortunately re-enabling SIP disables the driver. 
hope that helps