360Controller/360Controller #1152: Big Sur 11.2.3 / Apple M1 Chip / XBox 360 Wired Guitar

angeloliu1200, 2021-03-19:
### Type of Controller
Wired Xbox 360

### OS Version
Big Sur 11.2.3

### Driver Version
All versions of the drivers have been tested along with multiple debuggings from previous posts

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Macbook Air M1 2020
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Driver fails to install, but am still able to access the driver from the system preferences. 

### Response in Preference Pane
No devices found is the response
Cannot enable the driver as well

### Detailed Explaination of Issue
Currently, I have the driver 1.0.0-alpha.6 installed but I cannot enable the driver under Advanced. Also, the driver does not detect the controller with the message - No devices found
I have tried debugging with the links below but none have worked:
https://github.com/360Controller/360Controller/issues/1139
https://github.com/chris1111/Kext-Droplet-Big-Sur/releases/tag/V2

guylyons, 2021-04-01:
Having the same issue.

kailous, 2021-04-07:
Having the same issue.

softwaredeveloptam, 2021-05-02:
Same issue

**System**
Mac Mini M1

OS
macOS 11.3

adt-hgkim, 2021-05-12:
Same here.

**System**
MacBook Air M1

**OS**
macOS 11.3.1