360Controller/360Controller #1154: Razer Wolverine (Ultimate Edition) seems to be incompatible

beeters, 2021-03-20:
### Type of Controller
Xbox One

### OS Version
Catalina? I think so.

### Driver Version
The one that ends with "6."
(Newest one as of the time of typing this)

### Connection Method
Wired only
(Micro USB; I have to use a USB-C adaptor for my computer. Not a 3rd party one, the Apple one.)

### Device Name and Info
>Device Name: Razer Wolverine Ultimate Edition (Xbox One controller)
<br/>Product ID: NA
<br/>Vendor ID: NA

### Response of Controller
Plugging in the controller and connecting it causes the lights to turn/stay on, and there's the usual rumbling effect that happens whenever I turn it on. Everything normal in this part. However, ...

### Response in Preference Pane
None of the buttons, joysticks, and triggers have any input, as far as I know how to explain it; it thinks as if I'm not pressing any buttons/sticks/triggers, even though I am.

### Detailed Explaination of Issue
As far as I've been able to tell, the software seems to be incompatible for Razer Wolverine (ultimate edition) controller. I can connect it, but none of the buttons/sticks/triggers/etc. will function at all. Idk how to easily explain what I mean, but when it's connected in the "testing" menu, no inputs register anything with the computer/software; it thinks I'm not pressing a button or moving a joystick even when I am.

beeters, 2021-03-21:
I will come back to this periodically to check for replies, I won't abandon this like you said many many people did. Please get back to me as soon as you can. I searched my controller's name after making this post and I found that other people posted about this before me (but abandoned their posts), so I really believe the issue lies with the controller not being compatible with the driver. Please fix this as soon as you can

bblinder62, 2021-06-13:
I am having the same issue with the Wolverine Ultimate. Still no solution I could find. I'm just going to return my controller, good luck to you.

beeters, 2021-08-13:
> I am having the same issue with the Wolverine Ultimate. Still no solution I could find. I'm just going to return my controller, good luck to you.

Funny how the people who run this specifically instruct not to abandon the post or it'll be removed for inactivity, but then they don't even bother to address these issues.

Neutrinobeam, 2021-11-14:
The Razer Wolverine Ultimate Edition for Xbox One requires a different initialization packet when it is connected (detailed [here](https://github.com/quantus/xbox-one-controller-protocol).)  I've gotten it to work under MacOS 10.13.6 (including with Steam) by following these steps.

1. Remove the USB cord from the computer if the gamepad is already connected.
2. Open the Xbox 360 Controllers preference pane.
3. Under the Advanced Tab, uncheck "Enable Driver", authenticate as needed. Leave the preference pane open.
4. Connect the gamepad by USB, note the gamepad does not rumble nor light-up.
5. Set the gamepad to the default configuration using a third-party USB program (I use the Python3 shell and PyUSB 1.0 library). 
6. Observe the gamepad now rumbles and lights-up.
7. Write the packet [5,32,0,1,0] to the Out endpoint of the configuration's first interface using the same USB program.
8. Close the third-party USB program.
9. Return to the preference pane, recheck "Enable Driver", authenticating as needed.
10. The preference pane recognizes the gamepad as "Xbox One Wired Controller (Wired)" and responds to inputs.

One thing of note is that the Wolverine does not conform to the Elite gamepad protocol; the Macro Buttons are processed on the gamepad, by default emulating the A,B,X,Y set and altering the sticks' responsiveness. I have not attempted to reconfigure them or the LEDs.

Hipst3rMusic, 2023-07-01:
Hey Neutrino, I'm not very familiar with code/shell interfaces and I wanted to follow your method but I can't quite figure out step 5 - I've gotten python3 and PyUSB installed but don't know how to reset the controller to its default config. Could really use your help with step 5 & 7, thanks