360Controller/360Controller #1155: Controller connects to mac but buttons don’t work.

piperparis, 2021-03-24:
Wireless Xbox one controller

mac OS mojave Version 10.14.6

360Controller 1.0.0- alpha. 6 Driver Version

### Connection Method:
Controller can connect via bluetooth and Usb

### Response of Controller
I’m able to connect my controller to play fortnite but all the buttons are wrong. under system preferences: xbox 360 controllers: controller test it says no devices found. i allowed the security for drew mills when installing the driver and that still hasn’t enable access. i enabled the driver and i tried to adjust spoofing preferences by checking pretend to be an xbox 360 controllers but it won’t allow me to check the box. it appears dark so i can’t click on anything. I am able to use the controller to play fortnite however all the buttons are incorrect. A correlates to B, X to Y and so on. I have read/watched so many articles and videos and can’t figure out what i’m doing wrong. I’m just a girl tryna play some fortnite please help ///: it’s so frustrating

beeters, 2021-03-25:
I have the exact same issue. My controller is a Razer Wolverine Ultimate Edition. Only some specific third party brands seem to be compatible with this driver. I posted my issue days ago and idk how often they check these, so you an I might be sittin here for a while.

piperparis, 2021-03-25:
Have any idea what kind does work? i tried playing with an older wired controller and could never get that one to work but my new bluetooth one connects but buttons spaz out. i read somewhere the power A aren’t great with mac but idk what will work since the controller i’m using now is brand new

alexfanshawebabylon, 2021-03-26:
are you suffering the same version problem identified here: https://github.com/360Controller/360Controller/issues/1148

beeters, 2021-03-28:
> Have any idea what kind does work? i tried playing with an older wired controller and could never get that one to work but my new bluetooth one connects but buttons spaz out. i read somewhere the power A aren’t great with mac but idk what will work since the controller i’m using now is brand new

Hey, finally able to reply to you - I don't know which third party brands work, I'd do some searching in these threads. I remember reading somewhere on here that only "a few" third parties work. I'm sure somewhere in these threads there's a list of what brands and/or controller models work with this driver. If not, you can definitely look through the solved issues to find which ones work that way, but keep in mind that closed doesn't always mean solved because issues that were abandoned by the author for a long time got closed for inactivity.

L0VEFIRE, 2021-03-30:
this is in my post, the xbox one controller and newer work and must be bluetooth capable. the wired connections do not work anymore on Big Sur only Catalina. you have to use wireless on Big Sur; some ppl may get a rare fluke of luck where It works but out of my endless attempts never had 1 out of 40.

piperparis, 2021-03-30:
I have a brand new robot white xbox one wireless controller. bluetooth works but and connects to my computer but the buttons are all wrong. On fortnite i can’t build whatsoever or pull up my inventory. I get by but all the buttons are wrong. X goes to A. B to Y. RT to RB. all wrong

L0VEFIRE, 2021-03-30:
> I have a brand new robot white xbox one wireless controller. bluetooth works but and connects to my computer but the buttons are all wrong. On fortnite i can’t build whatsoever or pull up my inventory. I get by but all the buttons are wrong. X goes to A. B to Y. RT to RB. all wrong

ahh fortnite. well that is a epic games thing and they arnt easy to get them to fix stuff. just gonna have to go the the support forums and tell fortnite devs to fix there game on Mac, controller mappings are all wrong. and be persistent. 

no much you can do if you using Big Sur you ant using a wired connection it won't work. so either install bootcamp with windows 10 and play fortnite, get a PC, or find a way to install Catalina on a separate Mac Partition using disk utility( given you have enough space for that).

best you gonna get now.

beeters, 2021-03-31:
> this is in my post, the xbox one controller and newer work and must be bluetooth capable. the wired connections do not work anymore on Big Sur only Catalina. you have to use wireless on Big Sur; [...]

I am on Catalina and my wired controller does not work.

L0VEFIRE, 2021-04-01:
@beeters 

if your using catalina still, it should work. and you dont even need that stuff in this fourm well you shouldn’t. you need to make sure tho the driver get approved the old way in privacy and security pane. i found a lot of the time the driver install doesn’t initiate it so you just have to reinstall it again and make sure you have system preferences open main widow thwn install the deivee. when it give you the promot it need to be approved go to security and privacy pane and click the allow button at the buttom (hoping its there lol) should work. 

also i assume your using alpha 6 driver version

beeters, 2021-04-03:
> @beeters
>
> if your using catalina still, it should work. and you dont even need that stuff in this fourm well you shouldn’t. you need to make sure tho the driver get approved the old way in privacy [...]

No, no. The driver works fine with any other controller, my issue is that the third party controller I use is incompatible with the driver. I tested the driver (yes, the newest version) with 2 Microsoft Xbox One controllers (one with the headphone jack and one without), and both worked perfectly. Unfortunately, apparently only "a few" third party brands are compatible with this driver. Idk if these posts ever get read and reviewed anymore, but even if they did, I have to wonder if the driver ever gets updated to work with other brands. I keep reading in these threads something about the driver not being updated anymore.